Sapd/HeadsetControl #166: Added Roccat Elo 7.1 USB

JackLeEmmerdeur, 2021-07-19:
Hey folks. I've added support for the Elo USB headsets "led on/off" function by using the Elo Air code as template.
The Elo USB commands differ in just a few byte sequences. The buffer size of the data for send_receive() is 16 instead of 64 bytes. With 64 bytes the leds switched but the send_receive() timed out.

Is that device code wanted for the project and if yes should I make a pull-request and should it be in a seperate branch?

Sapd, 2021-08-30:
Hello,

of course you are free to make a pull request any time. Simply fork the project, then work on a branch you like (as its your fork it doesnt matter), then you can create a pull request. If something is off or so I will comment on that.

JackLeEmmerdeur, 2021-09-12:
See #172 for my PR. LED toggle was the only feature I was interested in as Surround/Stereo can be toggled from the linux driver. Thanks a lot for your application btw.

JackLeEmmerdeur, 2021-09-12:
Sorry. I somehow managed to commit with my alt-account. Can you delete the PR, so that I could try again with my main github user?

Sapd, 2021-09-12:
@JackLeEmmerdeur You can simply rebase it, and then do a `git push --force`. It will then overwrite it: https://stackoverflow.com/questions/3042437/how-to-change-the-commit-author-for-one-specific-commit

JackLeEmmerdeur, 2021-09-13:
I tried the most upvoted answer but at "git filter-branch -- --all" a whole lot of commits from others were rewritten.
I then tried the rebase/pick/edit way. Sadly my branch contains 2 commits and I only could manage to change one of the commits with the latter method. 
So in order to keep it easy I'm ok with the username in my proposed PR.
In all honesty it doesn't even matter if I none of my usernames is listed as a contributor at all and I'm just happy if this contribution saves anyone the hassle buying new headphones for linux.

JackLeEmmerdeur, 2021-09-26:
Hi @Sapd. 

As you probably can tell by the past conversation with me I'm not exactly well versed with github workflows, so I don't know how to proceed with the linter-error preventing a successful build. 

I understand from the diff, that it wants me to split the single-line "if" into two lines.

I did that but don't know what's the next step. Commit and push to the branch? Will that update the PR?

Feel free to do the changes yourself if that's a better approach.

closed 2021-10-14 by Sapd