Sapd/HeadsetControl #167: Arctis Pro is not recognized

Talkar, 2021-07-30:
If i type `headsetcontrol` i just get a message saying `No supported headset found`.
My headset is a Steelseries Arctis Pro.

Talkar, 2021-08-03:
I should probably add, this is the wired version, with a DAC, that i have.

johnhamelink, 2021-08-10:
I also have a Steelseries Arctis Pro. `lsusb` shows the following:

```
lsusb | grep SteelSeries
Bus 005 Device 012: ID 1038:1294 SteelSeries ApS Arctis Pro Wireless
Bus 005 Device 008: ID 1038:1290 SteelSeries ApS Arctis Pro Wireless
```

The Arctis Pro udev rules don't match:

```
cat /etc/udev/rules.d/50-steelseries-arctis-*
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1038", ATTRS{idProduct}=="12ad", MODE="0666"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1038", ATTRS{idProduct}=="1260", MODE="0666"
SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1038", ATTRS{idProduct}=="1252", MODE="0666"
```

Talkar, 2021-08-11:
My lsusb looks slightly different 
`lsusb | grep SteelSeries
Bus 001 Device 008: ID 1038:1280 SteelSeries ApS SteelSeries GameDAC
Bus 001 Device 012: ID 1038:1283 SteelSeries ApS SteelSeries GameDAC
`

csmith, 2021-08-23:
@johnhamelink Somewhat confusingly, the Arctis Pro Wireless is a different product to the Arctis Pro.

I recently added support for the Arctis Pro Wireless (see commit 199c6e00fcb533ebff18d8669a7f8d02106c718c), but there hasn't been a release since then so it won't work unless you compile a version from source.

@Talkar It looks like headsetcontrol expects the Arctis Pros to use product ID 1252:

```c
#define ID_ARCTIS_PRO_2019 0x1252
```

I guess your device is a newer model and they've switched the IDs. If it still uses the same protocol as the older version and you're happy compiling your own version of headsetcontrol, you could try changing that define in `src/devices/steelseries_arctis_7.c` to `0x1280` and see if it works.

johnhamelink, 2021-08-25:
@csmith yup, the headsetcontrol-git package on Arch's AUR now works with my Arctis Pro Wireless headphones!

Entropy512, 2021-12-06:
I've got a GameDAC coming in tonight, I'll try to shark it later this week.

It looks like there are two different configurations of Arctis Pro - one that has a "basic" USB DAC included and is merely sold as Arctis Pro, and then one that is bundled with a GameDAC, marketed as Arctis Pro + GameDAC.

I'm guessing that only the non-GameDAC version of the DAC is currently supported.

Once supported (hopefully not too difficult), the GameDAC should probably be listed separately as its own device since it can be used with any wired headset.

Entropy512, 2021-12-12:
PID 0x1252 is probably for the basic ChatMix DAC that is included with the basic Arctis Pro and also the Arctis 5 from the looks of Steelseries' website.

GameDAC is 0x1280 for the HID endpoint.  Audio is at 0x1282, 1283, or 1284 depending on which mode the DAC is in.

Here are my notes from Sharking the interface:
SET_REPORT 2a xx - Surround off (0)/ on (1)

SET_REPORT 2e xx - EQ settings: flat (0), bass boost (1), reference (2), smiley (3), custom (4)

SET_REPORT 33 xxxxxx - custom EQ adjust, 40 bytes, 10 bands, 4 bytes each, signed little-endian, max adjustment appears to be +/- 0x05000000

SET_REPORT 27 xx - Gain high (2)/low (1)

SET_REPORT 43 xx - Output Speakers (1)/Stream (2)

SET_REPORT 47 xx xx xx xx - Stream output, 4 bytes, from 0x00 (max) to 0x14 (min), for Game, Chat, Aux, Mic

SET_REPORT 39 xx - Sidetone, 0 (off) to 3 (high)

SET_REPORT 37 00 00 xx xx - 16-bit signed little endian, 0 is default mic of 8, 10 is 0x0300, 9 is 0x0180, 7 is 0xfe80, min is 0xf580

SET_REPORT 85 xx - OLED brightness, min is 0x18, max is 0xf0, default is 0xc0

SET_REPORT 83 xx xx - OLED screen timeout, 32 bit little endian, min (1 min) is 0xea60, 5 min is 0x0493e0,
10 min is 0x0927c0, 60min is 0x36ee80, never is 0x0 - 0xea60 is 60000, so this appears to be milliseconds

SET_REPORT a3 - no args?  commit???  sent when changing surround settings but always no parameters

SET_REPORT ac - seems to be sent just prior to commit/save

SET_REPORT 09 - commit/save

Some of the surround settings (Cinema/Game/Studio for example) always cause a SET_REPORT of 0xa3 to be sent - but there's always no data which confuses the hell out of me.

closed 2023-06-15 by Sapd

henryju, 2023-06-15:
Hi, the device looks to be supported now (using bd8aed3620920728843c2e4f839a9d4bf71802b2), but none of the commands seem to work:

```
# ./headsetcontrol -?
Found SteelSeries Arctis (7/Pro)!
Supported capabilities:

* sidetone
* battery status
* lights
* inactive time
* chatmix

# sudo ./headsetcontrol -l 0
Found SteelSeries Arctis (7/Pro)!
Success!
(but lights are not turned off)

# sudo ./headsetcontrol -m
Found SteelSeries Arctis (7/Pro)!
Failed to set/request chatmix, because of timeout, try again.

# sudo ./headsetcontrol -b
Found SteelSeries Arctis (7/Pro)!
Failed to set/request battery status. Error: -1: Broken pipe
```

PR #300 (2023-06-15): Adds Device ID for SteelSeries Arctis Pro GameDAC
SteelSeries Arctis Pro with GameDAC has a different device ID than with no GameDAC.
Resolves #167.