Sapd/HeadsetControl #168: Single tilt on the volume knob sometimes changes the volume by multiple levels

Luc45, 2021-07-31:
Hello,

I'm having this issue where the volume knob in the side of my Logitech G533 doesn't control the volume in a linear, reliable way. If I try to tilt a single time on the volume knob, the reaction isn't very reliable. If I try to roll the volume knob with my finger multiple levels, it rarely changes the volume in a predictable way, sometimes going all the way down, or changing one or two levels only.

I suspect that if I could add some kind of **small throttle to the volume knob input** this could be fixed. Eg: Only obey one input every 100ms

https://user-images.githubusercontent.com/9341686/127722728-ff797d62-1b46-437d-a5e7-68fbb473d329.mp4

Luc45, 2021-07-31:
Actually, I think this is being controlled by https://wiki.archlinux.org/title/PulseAudio#Keyboard_volume_control on my system

Sapd, 2021-08-30:
Yep its not a HeadsetControl Problem. The volume knobs etc. present themselves as Keyboard to the OS. And the OS (in this case Linux with X11 must interpret them correctly).

closed 2021-08-30 by Sapd