Sapd/HeadsetControl #169: Logitech Gaming Headset G733 battery level is very differnt from official software

clara2g, 2021-08-09:
Logitech G Hub on Windows reports reaming battery is 18% but HeadsetControl reports 66%
Thank you in advance.

Greifent, 2021-12-17:
Confirm that this issue is present, windows shows 15%, this shows 47, and reading go ups and down: ex: from 69 to 70 then 65, if any additional information is needed let me know!

Greifent, 2021-12-20:
Update, the headphones start reproducing the sound of low battery when the program shows 50%, it goes then down very fast to around 10% and then i noticed the red LED blinking on the headphones, the issue of the value not being constant (going from 10 to 16% without it being in charge) still remanins

Sapd, 2021-12-20:
> reading go ups and down: 

That readings go up and down is normal. The implementation simply reads the voltage, and tries to guess the battery level using this information. This is probably more worse when the battery is low.

I guess Logitech simply averages the readings to not confuse the user. Or they may go even further and calibrate the battery.

If your readings are far off, then it could be that the G733 may use a different type of battery then the other ones, then this function: [static float estimate_battery_level(uint16_t voltage)](https://github.com/Sapd/HeadsetControl/blob/ea74d6a69fb6409a2f03e88254b79eac7114ee29/src/devices/logitech_g633_g933_935.c#L44) would have to be adjusted.

This is how the function looks like:
https://www.wolframalpha.com/input/?i=f%28x%29%3D3.7268473047*10%5E%28-9%29x%5E%284%29-0.00005605626214573775*x%5E%283%29%2B0.3156051902814949*x%5E%282%29-788.0937250298629*x%2B736315.3077118985

Greifent, 2021-12-21:
Yeah, I was looking at the source code the other day and I guessed that it was based on voltage, but unfortunately my c knowledge stops there, I'll try to see if I can make my readings more precise.
If I put a print in that function, can I see the value of the voltage printed in the terminal when I run it? Sorry if it doesn't make sense but I'm not familiar with c

Sapd, 2021-12-21:
Yes you can simply insert
`printf("%u\n", voltage);`

Or before calling the function https://github.com/Sapd/HeadsetControl/blob/ea74d6a69fb6409a2f03e88254b79eac7114ee29/src/devices/logitech_g633_g933_935.c#L93 you can print the voltage and the result of the function as comparison:

`printf("%u \t- %f\n", voltage, estimate_battery_level(voltage));`

To calibrate it, you can simply record which voltage is which percentage. But note that this may only useful for you, as every battery (even the same model) is different, and even while the battery ages, the curve could change. The exception would be when the battery of this model uses a different kind, which has completely other voltage ranges.

Greifent, 2021-12-21:
Thank you, I will try to do that later, I'll let you know what I find out

hselasky, 2022-02-28:
I have installed headsetcontrol-2.6, and the battery request is timing out. But switching the LED off works!

Here is a USB trace from the control endpoint:

```
12:32:56.138922 usbus0.6 SUBM-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=8,IVAL=0
 frame[0] WRITE 8 bytes
 0000  80 06 00 03 00 00 40 00  -- -- -- -- -- -- -- --  |......@.        |
 frame[1] READ 64 bytes
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xeb9a3 <OPEN|TRANSFERRING|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CURR_DMA_SET|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.140525 usbus0.6 DONE-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=4,IVAL=0,ERR=0
 frame[0] WRITE 8 bytes
 frame[1] READ 4 bytes
 0000  04 03 09 04 -- -- -- --  -- -- -- -- -- -- -- --  |....            |
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xeb9a1 <OPEN|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CURR_DMA_SET|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.140567 usbus0.6 SUBM-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=8,IVAL=0
 frame[0] WRITE 8 bytes
 0000  80 06 00 03 00 00 40 00  -- -- -- -- -- -- -- --  |......@.        |
 frame[1] READ 64 bytes
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xcb9a3 <OPEN|TRANSFERRING|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.141645 usbus0.6 DONE-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=4,IVAL=0,ERR=0
 frame[0] WRITE 8 bytes
 frame[1] READ 4 bytes
 0000  04 03 09 04 -- -- -- --  -- -- -- -- -- -- -- --  |....            |
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xcb9a1 <OPEN|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.141686 usbus0.6 SUBM-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=8,IVAL=0
 frame[0] WRITE 8 bytes
 0000  80 06 03 03 09 04 00 02  -- -- -- -- -- -- -- --  |........        |
 frame[1] READ 512 bytes
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xeb9a3 <OPEN|TRANSFERRING|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CURR_DMA_SET|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.144146 usbus0.6 DONE-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=20,IVAL=0,ERR=0
 frame[0] WRITE 8 bytes
 frame[1] READ 18 bytes
 0000  12 03 4C 00 6F 00 67 00  69 00 74 00 65 00 63 00  |..L.o.g.i.t.e.c.|
 0010  68 00 -- -- -- -- -- --  -- -- -- -- -- -- -- --  |h.              |
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xeb9a1 <OPEN|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CURR_DMA_SET|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.144200 usbus0.6 SUBM-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=8,IVAL=0
 frame[0] WRITE 8 bytes
 0000  80 06 00 03 00 00 40 00  -- -- -- -- -- -- -- --  |......@.        |
 frame[1] READ 64 bytes
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xcb9a3 <OPEN|TRANSFERRING|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.145520 usbus0.6 DONE-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=4,IVAL=0,ERR=0
 frame[0] WRITE 8 bytes
 frame[1] READ 4 bytes
 0000  04 03 09 04 -- -- -- --  -- -- -- -- -- -- -- --  |....            |
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xcb9a1 <OPEN|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.145550 usbus0.6 SUBM-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=8,IVAL=0
 frame[0] WRITE 8 bytes
 0000  80 06 00 03 00 00 40 00  -- -- -- -- -- -- -- --  |......@.        |
 frame[1] READ 64 bytes
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xeb9a3 <OPEN|TRANSFERRING|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CURR_DMA_SET|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.146520 usbus0.6 DONE-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=4,IVAL=0,ERR=0
 frame[0] WRITE 8 bytes
 frame[1] READ 4 bytes
 0000  04 03 09 04 -- -- -- --  -- -- -- -- -- -- -- --  |....            |
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xeb9a1 <OPEN|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CURR_DMA_SET|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.146557 usbus0.6 SUBM-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=8,IVAL=0
 frame[0] WRITE 8 bytes
 0000  80 06 04 03 09 04 00 02  -- -- -- -- -- -- -- --  |........        |
 frame[1] READ 512 bytes
[... 75 lines cut ...]
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xcb9a3 <OPEN|TRANSFERRING|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.165919 usbus0.6 DONE-CTRL-EP=00000080,SPD=FULL,NFR=2,SLEN=40,IVAL=0,ERR=0
 frame[0] WRITE 8 bytes
 frame[1] READ 40 bytes
 0000  28 03 47 00 37 00 33 00  33 00 20 00 47 00 61 00  |(.G.7.3.3. .G.a.|
 0010  6D 00 69 00 6E 00 67 00  20 00 48 00 65 00 61 00  |m.i.n.g. .H.e.a.|
 0020  64 00 73 00 65 00 74 00  -- -- -- -- -- -- -- --  |d.s.e.t.        |
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xcb9a1 <OPEN|STARTED|CONTROL_XFR|CONTROL_HDR|SHORT_FRAMES_OK|SHORT_XFER_OK|BDMA_ENABLE|BDMA_SETUP|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.167989 usbus0.6 SUBM-CTRL-EP=00000000,SPD=FULL,NFR=2,SLEN=28,IVAL=0
 frame[0] WRITE 8 bytes
 0000  21 09 11 02 03 00 14 00  -- -- -- -- -- -- -- --  |!.......        |
 frame[1] WRITE 20 bytes
 0000  11 FF 08 0A 00 00 00 00  00 00 00 00 00 00 00 00  |................|
 0010  00 00 00 00 -- -- -- --  -- -- -- -- -- -- -- --  |....            |
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xea1a3 <OPEN|TRANSFERRING|STARTED|CONTROL_XFR|CONTROL_HDR|BDMA_ENABLE|BDMA_SETUP|CURR_DMA_SET|CAN_CANCEL_IMMED|DOING_CALLBACK|0>
12:32:56.168657 usbus0.6 DONE-CTRL-EP=00000000,SPD=FULL,NFR=2,SLEN=0,IVAL=0,ERR=0
 frame[0] WRITE 8 bytes
 frame[1] WRITE 20 bytes
 flags 0x2 <SHORT_XFER_OK|0>
 status 0xea1a1 <OPEN|STARTED|CONTROL_XFR|CONTROL_HDR|BDMA_ENABLE|BDMA_SETUP|CURR_DMA_SET|CAN_CANCEL_IMMED|DOING_CALLBACK|0>

```

hselasky, 2022-02-28:
Found the solution:

`sysctl hw.usb.uaudio.handle_hid=0
`
Looks like there is a conflict receiving HID packets on the interrupt endpoint.

```
headsetcontrol -b
Found Logitech G633/G635/G933/G935!
Battery: 91%
Success!

```

lezcano17, 2023-06-22:
Im facing this issue

TobiasS-jpg, 2024-03-24:
I have tinkered around in the past and, as Sapd already said, adjusting the
`static float estimate_battery_level(uint16_t voltage)`
function is the solution.

This code works for me:
```c
static float estimate_battery_level(uint16_t voltage)
{
    // Charging
    if (voltage == 65535)
        return (float) (BATTERY_CHARGING);

    // Disconnected
    if (voltage <= 3000)
        return (float) (BATTERY_UNAVAILABLE);

    // Empty
    if (voltage <= 3647)
        return (float) 0.0;

    // Polynom 1
    if (voltage <= 3726)
        return (float) (0.0000671638202 * pow(voltage, 3) - 0.746307678145 * pow(voltage, 2) + 2764.3173240131123 * voltage - 3413055.137837838);

    // Polynom 2
    if (voltage <= 3885)
        return (float) (-0.0000112590873 * pow(voltage, 3) + 0.1272931815966 * pow(voltage, 2) - 479.336522317864 * voltage + 601218.8751422882);

    // Polynom 3
    if (voltage <= 3984)
        return (float) (-0.000002711158 * pow(voltage, 3) + 0.032582506827 * pow(voltage, 2) - 130.3351489423122 * voltage + 173611.6066702439);

    // Polynom 4
    if (voltage <= 4120)
        return (float) (0.000010859959 * pow(voltage, 3) - 0.131425350889 * pow(voltage, 2) + 530.2926489312302 * voltage - 713325.3412124208);

    return (float) 100.0;
}
```

I came up with these values by simply periodically checking the voltage with HeadsetControl and then rebooting into Windows to check the according percentage in G HUB.

closed 2025-03-20 by github-actions[bot]