Sapd/HeadsetControl #170: Device Request: Corsair Virtuoso Non-SE

Anigai, 2021-08-15:
Hi,

I'd like to request the adding of the device Corsair Virtuoso standard edition(plastic ear muffs, not aluminium).
The main feature I'd like to see is the disabling of the RGB lights which I've noticed drain it's battery very quickly on Linux compared to my turning them off in the available Windows iCUE software.

I'm a little out of my depth after reading the Development Wiki in such that I understand the concept of retrieving and copying the same USB traffic to the Wireless headset via the 2.4ghz USB dongle but I don't believe I can go through the process alone and be successful without more knowledge.

For reference my level of Linux knowledge is about just below the requirements to pass the CompTia Linux+ exam but I've got a Raspberry Pi setup as a NAS using OpenMediaVault as well as some knowledge of Docker.

Can you guys provide any recommendations?

Cheers

polarnixdev, 2021-09-24:
+1 I would like support for this device. 

I was able to get sidetone working for this device using alsamixer and it works flawlessly. 

I also tried following the instructions to add a corsair device to HeadsetControl and no luck. It says "Found Corsair Headset Device!" followed by "Couldn't open device.".

Setherz6104, 2022-08-14:
+1 I LOVE my Virtuoso!

Setherz6104, 2022-08-14:
@xinral How were you able to get sidetone to work with AslaMixer? I opened AlsaMixer and found my Virtuoso; I'm able to adjust the Mic and Headset values fine, but increasing Sidetone isn't doing anything

xinral, 2022-08-14:
@Setherz6104 In Alsa you can press "f5" or something similar to view all options for the headset, there _should_ be an option for sidetone, not sure what its called off the top of my head because I dont use this headset anymore. There arent too many options in alsa so if you dig around you should find it.

There are also different versions of this headset so maybe not all of them have this option in alsamixer? I cant say for sure.

Setherz6104, 2022-08-14:
> @Setherz6104 In Alsa you can press "f5" or something similar to view all options for the headset, there _should_ be an option for sidetone, not sure what its called off the top of my head because I [...]

No I see Sidetone as an option, what I mean is increasing the slider doesn't do anything, the Sidetone just stays quiet like usual.

xinral, 2022-08-14:
@Setherz6104 my bad for misreading, this worked for me. make sure your sidetone is actually enabled on the headset, you might have to enable it in windows before booting linux unless theres a physical button to turn it on/off.

Setherz6104, 2022-08-14:
> @Setherz6104 my bad for misreading, this worked for me. make sure your sidetone is actually enabled on the headset, you might have to enable it in windows before booting linux unless theres a physical button to turn it on/off. 

At least for my model, there's no physical button for Sidetone and iCUE doesn't save ANY changes to hardware as a profile; iCUE has to be running in Windows to set Sidetone, EQ, etc.

It's too bad cause Sidetone is one of the main things I look for in a headset

s-thoeni, 2022-08-14:
I got it working on mine by long-pressing the mute button of the microphone.

Anigai, 2022-08-14:
For reference, I returned this set of headphones due to eventually deciding the shape of them didn't agree with my head as much as my previous Kingston HyperX Cloud II's so I instead bought a Kingston HyperX Cloud II Wireless and while they're a bit more basic than the Corsairs I think they're much better for me.

Feel free to close out this request though I imagine some people will still want this feature.

Setherz6104, 2022-08-15:
> 

YOU ARE MY SAVIOR!!!!!!