Sapd/HeadsetControl #177: Device Request: CORSAIR VIRTUOSO XT Wireless

echo2go, 2021-10-14:
I have tried to compile your code for "CORSAIR VIRTUOSO XT Wireless" but it failed.
Maybe you can help to add this headset to your code?

`
== dmesg ==

[  659.853696] usb 1-2: new full-speed USB device number 12 using xhci_hcd
[  660.002805] usb 1-2: New USB device found, idVendor=1b1c, idProduct=0a64, bcdDevice= 0.00
[  660.002811] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  660.002815] usb 1-2: Product: CORSAIR VIRTUOSO XT Wireless Gaming Receiver
[  660.002818] usb 1-2: Manufacturer: Corsair
[  660.002821] usb 1-2: SerialNumber: 167d7765000700da
[  660.265180] input: Corsair CORSAIR VIRTUOSO XT Wireless Gaming Receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.3/0003:1B1C:0A64.0007/input/input31
[  660.320857] input: Corsair CORSAIR VIRTUOSO XT Wireless Gaming Receiver Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.3/0003:1B1C:0A64.0007/input/input32
[  660.321137] input: Corsair CORSAIR VIRTUOSO XT Wireless Gaming Receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.3/0003:1B1C:0A64.0007/input/input33
[  660.321403] input: Corsair CORSAIR VIRTUOSO XT Wireless Gaming Receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.3/0003:1B1C:0A64.0007/input/input34
[  660.321661] input: Corsair CORSAIR VIRTUOSO XT Wireless Gaming Receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.3/0003:1B1C:0A64.0007/input/input35
[  660.322167] hid-generic 0003:1B1C:0A64.0007: input,hiddev96,hidraw0: USB HID v1.11 Device [Corsair CORSAIR VIRTUOSO XT Wireless Gaming Receiver] on usb-0000:00:14.0-2/input3
[  660.324282] hid-generic 0003:1B1C:0A64.0008: hiddev97,hidraw1: USB HID v1.11 Device [Corsair CORSAIR VIRTUOSO XT Wireless Gaming Receiver] on usb-0000:00:14.0-2/input4

== dmesg END ==

~/HeadsetControl-master $ lsusb | grep -i corsair
Bus 001 Device 012: ID 1b1c:0a64 Corsair CORSAIR VIRTUOSO XT Wireless Gaming Receiver
~/HeadsetControl-master $ 
~/HeadsetControl-master $ 
~/HeadsetControl-master $ cat /etc/udev/rules.d/headsets.rules 
# CORSAIR VIRTUOSO XT Wireless Gaming Receiver
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0a64", TAG+="uaccess"
#SUBSYSTEM=="hidraw", SUBSYSTEM=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0a64", MODE="0666"
~/HeadsetControl-master $ 
~/HeadsetControl-master $ 
~/HeadsetControl-master $ grep -ni virtuoso src/devices/corsair_void.c 
27:#define ID_CORSAIR_VIRTUOSO_XT_Wireless_Gaming_Receiver    0x0a64
44:    ID_CORSAIR_VIRTUOSO_XT_Wireless_Gaming_Receiver
~/HeadsetControl-master $ 
~/HeadsetControl-master $ 
~/HeadsetControl-master $ ./build/headsetcontrol -b
Found Corsair Headset Device!

Couldn't open device.
~/HeadsetControl-master $
~/HeadsetControl-master $
`

LeonardoNozaki, 2021-11-30:
I'm trying to use HeadsetControl with Corsair Virtuoso SE, but something wrong :(

I'm using Ubuntu 20.04, I tested with Corsair Virtuoso SE USB and Corsair Virtuoso SE Wifi, but both returns **"No supported headset found"**

Any suggestion?

```
~  HeadsetControl git:(master) sudo whoami
root
~
~  HeadsetControl git:(master) sudo lsusb | grep -i corsair 
Bus 001 Device 011: ID 1b1c:0a3f Corsair CORSAIR VIRTUOSO SE USB Gaming Headset
Bus 001 Device 008: ID 258a:002a Corsair CORSAIR VIRTUOSO SE USB Gaming Headset
~
~  HeadsetControl git:(master) sudo cat /etc/udev/rules.d/headsets.rules
cat: /etc/udev/rules.d/headsets.rules: No such file or directory
~  HeadsetControl git:(master) sudo grep -ni virtuoso src/devices/corsair_void.c
~  HeadsetControl git:(master) sudo ./build/headsetcontrol -b
No supported headset found
~  HeadsetControl git:(master) sudo time ./build/headsetcontrol -b
No supported headset found
Command exited with non-zero status 1
0.00user 0.01system 0:00.01elapsed 92%CPU (0avgtext+0avgdata 2672maxresident)k
0inputs+0outputs (0major+117minor)pagefaults 0swaps
~  HeadsetControl git:(master) sudo ./build/headsetcontrol -h
Headsetcontrol written by Sapd (Denis Arnst)
	https://github.com/Sapd

Parameters
  -s, --sidetone level		Sets sidetone, level must be between 0 and 128
  -b, --battery			Checks the battery level
  -n, --notificate soundid	Makes the headset play a notifiation
  -l, --light 0|1		Switch lights (0 = off, 1 = on)
  -c, --short-output		Use more machine-friendly output 
  -i, --inactive-time time	Sets inactive time in minutes, time must be between 0 and 90, 0 disables the feature.
  -m, --chatmix			Retrieves the current chat-mix-dial level setting between 0 and 128. Below 64 is the game side and above is the chat side.
  -v, --voice-prompt 0|1	Turn voice prompts on or off (0 = off, 1 = on)
  -r, --rotate-to-mute 0|1	Turn rotate to mute feature on or off (0 = off, 1 = on)
  -?, --capabilities		Print every feature headsetcontrol supports of the connected headset

  -u	Outputs udev rules to stdout/console

~  HeadsetControl git:(master)
```

echo2go, 2022-04-07:
I have found https://github.com/Sapd/HeadsetControl/wiki/Development#adding-a-new-headset and try to capture the data.
First I download USBlyzer and installed in Windows 10.
Then looked for the sidetone slicer and play around it.
I have made a screenshot and a dump of the trace. Hope it helps.

[image]
[corsair_virtuoso_xt_tracert.zip](https://github.com/Sapd/HeadsetControl/files/8445053/corsair_virtuoso_xt_tracert.zip)

migmolrod, 2022-10-01:
I'm also trying the Corsair Virtuoso XT in Wireless mode, device id = 0x0a64 (in USB mode it's 0x0a63, I think).
```
> grep -ni virtuoso src/devices/corsair_void.c
27:#define ID_CORSAIR_VIRTUOSO_XT_RECEIVER    0x0a64
44:    ID_CORSAIR_VIRTUOSO_XT_RECEIVER,
```

What I've tried:
Just `headsetcontrol` (after `make install`, ie, using `/usr/local/bin/headsetcontrol`), without and with sudo:
```
> headsetcontrol -b
Found Corsair Headset Device!
Failed to set/request battery status, because of timeout, try again.

> sudo !!
sudo headsetcontrol -b
Found Corsair Headset Device!
Failed to set/request battery status, because of timeout, try again.
```
Using the built script (also without and with sudo)
```
> ./build/headsetcontrol -b
Found Corsair Headset Device!
Failed to set/request battery status, because of timeout, try again.

> sudo !!
sudo ./build/headsetcontrol -b
Found Corsair Headset Device!
Failed to set/request battery status, because of timeout, try again.
```

I suppose these have a different protocol or internal interfaces or whatever, right?

rsimai, 2022-10-06:
FWIW seeing exactly the same. I'd love to at least read the battery.

rsimai, 2022-10-13:
for the sake of sharing pain and if anyone wants to join the "constant dropping" party, I approached Corsair support to ask for the protocols and they have been responsive, but finally received the expected:

```
Unfortunately we can not refer you any further in this matter.
Linux is not supported by any of our software in any capacity at this time.
We have already reached out to the engineering team to find out if the requested information can be provided -- this request was denied.

We appreciate your patience, but unfortunately this is simply something that we are not able to assist you with.
```

closed 2023-10-10 by github-actions[bot]