Sapd/HeadsetControl #182: Arctis 9 support on Win release

Jakusashi, 2021-11-02:
I recently bought a SS Arctis 9 headset and first thing it comes to my mind was "it would be nice to see exactly how many battery % left". That's why I downloaded firstly 2.5, then 2.4 version for Windows.
Application recognize headset after command "headsetcontrol.exe -b", and prints "Found SteelSeries Arctis 9!", but nothing else. It looks like it loops, so I have to break it with ctrl+c.

Headset software is of course up-to-date and it doesn't matter whether I use it wirelessly or charging via USB cable. 
I downloaded WireShark so if U need any IDs, I'll gladly help

Sapd, 2021-11-02:
Does Sidetone work?

If yes, the problem is most likely, that the battery feature is in a different usagepageid.
That means the software can currently only support one feature at a time, as it wasn't developed with that in mind (as on Linux, it does not matter which usage-page you use, as there is none). So unfortunately you can only use one feature without recompiling the software. (See https://github.com/Sapd/HeadsetControl/issues/154 for more details)

In fact I already have an idea how to solve this problem (which exists on Windows). However it needs bigger changes in code, where I can not yet provide an ETA.

Jakusashi, 2021-11-02:
Yes, sidetone works fine. I'll try to recompile the software for battery status

Sapd, 2021-11-13:
> That means the software can currently only support one feature at a time, as it wasn't developed with that in mind (as on Linux, it does not matter which usage-page you use, as there is none). So [...]

This is completely implemented with the new version. However you may need to add the correct usageID in the source code (like in corsairs_void.c).

`./headsetcontrol --dev -- --list` will list all devices with their usageIDs, so you can simply brute force through them

Jakusashi, 2021-11-14:
Oh, nice. Thank you. 
It looks like _0x1_ is a usageID of this headset. Currently I don't have much time, but I'll try to add this ID to code and recompile it.
I'll let you know when everything works

Sapd, 2021-11-14:
You can also test it without recompiling now, e.g.:
`./headsetcontrol --dev -- --device 0x1038:0x12c2 --usage 0x0:0x1 --send "0x20, 0x00" --sleep 50 --receive`

and if it works add it later to code and PR

Jakusashi, 2021-11-14:
I tried this command and unfortunatelly it returns _No data to read_, when I'm used `--usage 0x0:0x1`, `--usage 0xff00:0x1` and `--usage 0xffc0:0x1` or _Failed to send data. Error -1_ with `--usage 0xc:0x1` (0xff00, 0xffc0 and 0xc is a different usage pages).
I'm not sure if I should change something else

Sapd, 2021-11-14:
Hmm "No data to read" sounds however like the request is correct. Maybe simple try it multiple times, maybe also without the --sleep option.
Otherwise you have to indeed try it in code, could be a problem with the --dev interface - as it is relatively new, even when I tested it.

Karpovich-alex, 2022-04-07:
Hi there I also faced this issue with Arctis 9 on win10.
After the command "headsetcontrol.exe -b", it prints "Found SteelSeries Arctis 9!", but nothing else.
but if you open SS GG software, go to the engine and _change tabs from gear to apps_ (or other tabs), the script works fine and displays information about the battery.

Also this script works, if you change tabs in SS GG app in the meantime.
`./headsetcontrol --dev -- --device 0x1038:0x12c2 --usage 0x0:0x1 --send "0x20, 0x00" --sleep 500 --receive`

I dont know how it is connected, but other scripts [(example js script)](https://github.com/atagulalan/arctis-battery-percentage) only work for me when switching tabs in the SS GG application.

Karpovich-alex, 2022-04-25:
Hi there. I think I found out why it doesn't work.
I found that when you switch tabs in the SteelSeries GG application, it sends HID SET_REPOST request with Data Fragment = 200...000 and after that URB_INTERRUPT in, in response to which it receives HID Data with information about your headset.
In the meantime, the script sends a HID SET_REPOST request with Data Fragment = 000...000  which does not produce URB_INTERRUPT packet (sorry for my terms, I'm not well versed in the HID interface)
So if you will send similar request as SS GG using script
`./headsetcontrol --dev -- --device 0x1038:0x12c2 --usage 0x0:0x1 --send "0x20, 0x20" --sleep 500 --receive`
it should return the correct information about the state of your battery.
@Jakusashi can you test this code with your headset and let me know, does this method works with your Arctis.

Toomoch, 2022-08-02:
> 

Hi, I just tried the line you sent with my arctis 9, and this is the output:

`0xaa 0x01 0x01 0x9c 0000 0000 0000 0000 0xc0 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000`

Karpovich-alex, 2022-08-24:
@Toomoch 
Hi, I've created a pull request with fixing this problem. I think the problem was in the parameters of the request for information about the status of the headphones

closed 2023-05-01 by Sapd