Sapd/HeadsetControl #193: Battery status freeze

bioxz, 2021-11-17:
### Description

When using headsetcontrol on a regular basis (e.g. in scripts) it tends to freeze after the device name output or directly when using the short output after a some executions. This was already reported in #125, but doesn't seem to be limited to a specific device, in my case it's a SteelSeries Arctis 9.  Instead of

>Found SteelSeries Arctis 9!
>Battery: 31%
>Success!

I just get a

>Found SteelSeries Arctis 9!

and no further output. Sometimes this happens after the second or third execution of headsetcontrol, sometimes it works for 100 executions without a problem.

This makes projects like the mentioned [headsetcontrol-notificationd](https://github.com/Manawyrm/headsetcontrol-notificationd) unusable, as the script locks up in the background.

If one instance of headsetcontrol hangs after requesting the battery status, all other instances querying the battery status will freeze as well. Terminating all instances and waiting for a bit seems to solve the problem and a new instance of headsetcontrol will display the battery status again -- until a future request will freeze again.

System information:

> Debian stable
> Linux 5.10.70
> udev 247.3
> libhidapi 0.10.1

### On which OS does the problem happen?

Linux

Sapd, 2021-11-17:
Possibly HID hangs then because it is waiting for data, but the headset doesn't provide it.

We can easily fix it by adding a timeout (maybe even with a --timeout) parameter

bioxz, 2021-11-17:
That sounds like a good idea. Please let me know if I can provide more specific debug information or test an implementation.

Sapd, 2021-11-19:
Will probably work on it next week (lockdown announced anyway...)

Sapd, 2021-11-24:
You can try the branch `timeout-fix` now
E.g.
`git pull`
`git checkout timeout-fix`

then delete the build folder and rebuild.

Default is 5 secounds, you can change it with the new `--timeout millisecounds` parameter.
On timeout it will return an error, it may need to be handled by the notification application you use.

PR #195 (2021-12-03): Added timeout parameter for reading, default is 5s
Also:
- Improve parameters with numbers, check if they are really numbers
- Pass error codes to program return
- On timeout, program returns -101

Fixed #193
closed 2021-12-03 by Sapd