Sapd/HeadsetControl #194: Logitech Zone Wired setting sidetone disables it

culebrasolida, 2021-11-18:
### Description

Whenever I try to set sidetone volume for a Logitech Zone Wired headset with:
`sudo headsetcontrol -s --50
Found Logitech Zone Wired/Zone 750!
Success!
`

That completely disables sidetone and headset has to be plugged back in to get sidetone back.

### Headset Name

Logitech Zone Wired

### On which OS does the problem happen?

Linux

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
Device Found
  VendorID: 0x046d
 ProductID: 0x0aad
  path: /dev/hidraw3
  serial_number: 2017ALQ00SV2
  Manufacturer: Logitech
  Product:      Zone Wired
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x046d
 ProductID: 0x0aad
  path: /dev/hidraw3
  serial_number: 2017ALQ00SV2
  Manufacturer: Logitech
  Product:      Zone Wired
  Interface:    3
  Usage-Page: 0xff99 Usageid: 0x4

Device Found
  VendorID: 0x046d
 ProductID: 0x0aad
  path: /dev/hidraw3
  serial_number: 2017ALQ00SV2
  Manufacturer: Logitech
  Product:      Zone Wired
  Interface:    3
  Usage-Page: 0xb Usageid: 0x5

Device Found
  VendorID: 0x046d
 ProductID: 0x0aad
  path: /dev/hidraw3
  serial_number: 2017ALQ00SV2
  Manufacturer: Logitech
  Product:      Zone Wired
  Interface:    3
  Usage-Page: 0xb Usageid: 0x6

Device Found
  VendorID: 0x046d
 ProductID: 0x0aad
  path: /dev/hidraw3
  serial_number: 2017ALQ00SV2
  Manufacturer: Logitech
  Product:      Zone Wired
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0x5

Device Found
  VendorID: 0x046d
 ProductID: 0x0aad
  path: /dev/hidraw3
  serial_number: 2017ALQ00SV2
  Manufacturer: Logitech
  Product:      Zone Wired
  Interface:    3
  Usage-Page: 0xff30 Usageid: 0x6

Device Found
  VendorID: 0x046d
 ProductID: 0x0aad
  path: /dev/hidraw3
  serial_number: 2017ALQ00SV2
  Manufacturer: Logitech
  Product:      Zone Wired
  Interface:    3
  Usage-Page: 0xff53 Usageid: 0x404

Device Found
  VendorID: 0x046d
 ProductID: 0x0aad
  path: /dev/hidraw3
  serial_number: 2017ALQ00SV2
  Manufacturer: Logitech
  Product:      Zone Wired
  Interface:    3
  Usage-Page: 0xff99 Usageid: 0x3
 ```

</details>

Sapd, 2021-11-18:
Is `sudo headsetcontrol -s --50` with **--50** what you actually used? What happens if you try it correctly with  `sudo headsetcontrol -s 50`?

Sapd, 2021-12-03:
I did better error parsing in the last commit, before something like --50 was silently truncated to 0. Now it gives an explicit error. 
(Error handling in C is a mess)

closed 2021-12-03 by Sapd