Sapd/HeadsetControl #213: Corsair Void Elite USB Sidetone not working

Biosha, 2022-05-09:
### Description

I tried to set the sidetone of my Corsair Void Elite USB but this message appears :

`headsetcontrol -s 0
Found Corsair Headset Device!
ioctl (SFEATURE): Broken pipe
Failed to set/request sidetone. Error: -1: (null)
`

### Headset Name

Corsair Void Elite USB

### On which OS does the problem happen?

Linux

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a52
  path: /dev/hidraw6
  serial_number: 00000000
  Manufacturer: Cosair
  Product:      Corsair VOID ELITE USB Gaming Headset
  Interface:    3
  Usage-Page: 0x9de9 Usageid: 0x4203
 ```

</details>

aDwCarrazzone, 2022-06-08:
Same thing here =/

Sapd, 2023-05-01:
Closing, the sidetone is controlled via an audio channel, which we can't access via HID. Try controlling it with alsamixer

closed 2023-05-01 by Sapd