Sapd/HeadsetControl #215: Is there a way to apply the Speex and/or RNNoise noise filter to sidetone?

BombJovi, 2022-05-22:
### Description

There's constant background noise in my room from my ceiling fan constantly blowing air into my headset, causing the side-tone driver to pick up the noise from the wind flow and I hear it. It would be nice to filter out the wind noise. Mumble has a great noise suppression filter where it doesn't pick up the wind noise because of Speex + RNNoise. Is there a way to filter out everything besides my voice for the HeadsetControl driver? That would be neat!

Thank you!

Sapd, 2022-05-24:
> Is there a way to filter out everything besides my voice for the HeadsetControl driver

The sidetone is usually transmitting directly from the mic to the speakers of the headset. It does not go through the PC first and then back, as it would create disturbing latency. So there is no way to add any kind of filter or such, the hardware of the headset must provide that by itself.

I would suggest just trying to cut a piece of foam and put it around the mic.

BombJovi, 2022-05-24:
Right on! Thank you Sapd for the reply and for making an awesome piece of software! I use headsetcontrol all the time. I'll close the issue and note your recommendation.

closed 2022-05-24 by BombJovi