Sapd/HeadsetControl #217: Releases updates?

wael444, 2022-05-26:
### Description

There are currently new commits and updates but it's not being updated/added via https://github.com/Sapd/HeadsetControl/releases.

Also this label is quite inaccurate but its the closest.

### On which OS does the problem happen?

Linux

Sapd, 2022-07-17:
Indeed  your Issue is quite valid. I wanted to plan this for a long time but I'm not sure if I should still keep tagging releases manually.

The problem is, for this kind of software a pull of a Release does not make much sense. In fact I'm currently using the releases to publish the Windows artefacts which are hard to build. In Winter I plan to automatically at least publish artefacts for that. Ultimate goal of course would be to have automated debian or yum packages, but given I have still limited knowledge for that it would take time.

Best way for automation scripts is to actually pull from master currently.

Sapd, 2022-07-17:
Will do a manual release later this day.

Also something like an automated package build would be probably also interesting for @centic9 s Appindicator.
However I will probably again resuming work on that in Winter

closed 2023-05-01 by Sapd

apprehensions, 2023-05-01:
2.6.2 latest tag from July of last year sar

Sapd, 2023-05-01:
Just pushed a new release. I can also do this now more frequently, as a Windows artefact is also now uploaded automatically by the CI.

Note that Linux and Mac users should probably always use the latest HEAD. I made a paragraph on that in the README: https://github.com/Sapd/HeadsetControl#release-cycle