Sapd/HeadsetControl #218: Arctis 7x Device Support

blades, 2022-05-27:
### Description

I can see we've got a 7 and a 7+, but it doesn't seem to recognise the 7x. 

I'll see what I can figure out, but if there are any pointers as to where to look and what to look at that would be awesome.

### Headset Name

Steelseries Arctis 7x

### Device information

<details>
 <summary>Detailed Device Information</summary>
Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff43 Usageid: 0x202

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xc

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0x1 Usageid: 0x3a

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xc

Device Found
  VendorID: 0x1038
 ProductID: 0x12d7
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 7X
  Interface:    3
  Usage-Page: 0x1 Usageid: 0x3a

</details>

blades, 2022-05-27:
Okay, so I've tried adding the ID 0x12d7 to the various steelseries classes, and can find the device but the battery level command just gives me 'Requested/supported HID device not found or system error. HID Error: Success' each time so I presume it's going to be different to each of the existing devices. Lovely. I've added new files to create a new one and I'll see if I can glean anything from wireguard, although it's fair to say that at this point I'm rapidly getting out of my depth :)

blades, 2022-06-01:
Evidently I made a mistake somewhere along the way because the arctis 1 class appears to work for the 7x as well. I'll push a change for that up.

Sapd, 2022-07-17:
Merged

closed 2022-07-17 by Sapd