Sapd/HeadsetControl #222: Ignore udev installation when compiling on FreeBSD

Alexander88207, 2022-07-24:
### Description

Hello,

First of all thank you for this tool, it works great.

I do maintain the [port on FreeBSD](https://www.freshports.org/audio/headsetcontrol/) and we dont use udev. Since i have no idea about cmake, i just removed this section that is installing udev files, maybe you have a better solution? So that it is no longer necessary from our side modifying the CMakeLists.txt endlessly.

Thank you in advance!

### On which OS does the problem happen?

Linux

Sapd, 2022-07-24:
Hello,
thank you for providing a port!

According to https://gitlab.kitware.com/cmake/community/-/wikis/doc/tutorials/How-To-Write-Platform-Checks#cmake-system

A variable `FreeBSD` exists.
So we probably just have to change the line https://github.com/Sapd/HeadsetControl/blob/d890a6cf45eb459e94fc3e6276f310ea49248765/CMakeLists.txt#L122
to
```cmake
if(UNIX AND NOT APPLE AND NOT ${CMAKE_HOST_SYSTEM_NAME} MATCHES "FreeBSD") 
```

if you could test it (or directly create a PR) we can include it

Alexander88207, 2022-07-24:
Thank you for your quick reply.

It works as expected, a PR was filled in.

PR #223 (2022-07-24): Also dont install udev files on FreeBSD
FreeBSD dont uses udev.

Fixes #222

Thank you!
closed 2022-07-24 by Sapd