Sapd/HeadsetControl #224: G435 Battery status

t0int, 2022-08-07:
### Description

Battery status. There is no software (I guess) that shows the battery right now.

### Headset Name

Logitech G435

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
 Device Found
  VendorID: 0x046d
 ProductID: 0x0acb
  path: \\?\HID#VID_046D&PID_0ACB&MI_03&Col03#7&64d5b1&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 202105190004
  Manufacturer: Logitech G series
  Product:      G435 Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc0 Usageid: 0x1

Device Found
  VendorID: 0x046d
 ProductID: 0x0acb
  path: \\?\HID#VID_046D&PID_0ACB&MI_03&Col01#7&64d5b1&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 202105190004
  Manufacturer: Logitech G series
  Product:      G435 Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x046d
 ProductID: 0x0acb
  path: \\?\HID#VID_046D&PID_0ACB&MI_03&Col04#7&64d5b1&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 202105190004
  Manufacturer: Logitech G series
  Product:      G435 Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc3 Usageid: 0x1

Device Found
  VendorID: 0x046d
 ProductID: 0x0acb
  path: \\?\HID#VID_046D&PID_0ACB&MI_03&Col02#7&64d5b1&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 202105190004
  Manufacturer: Logitech G series
  Product:      G435 Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xb Usageid: 0x1
 ```

</details>

Sapd, 2023-06-05:
I would recommend trying to use WireShark to intercept the packets of the logitech software: https://github.com/Sapd/HeadsetControl/wiki/Development#adding-a-completely-new-headset

(can be also done in a Virtual Machine). Bc. only people who own the headset can actually find out which kind of packets are needed.

t0int, 2023-06-22:
> I would recommend trying to use WireShark to intercept the packets of the logitech software: https://github.com/Sapd/HeadsetControl/wiki/Development#adding-a-completely-new-headset
>
> (can be also [...]

It doesn't have any software that will send packets.

t0int, 2023-06-22:
> I would recommend trying to use WireShark to intercept the packets of the logitech software: https://github.com/Sapd/HeadsetControl/wiki/Development#adding-a-completely-new-headset
>
> (can be also [...]

there is no logitech software for this device.