Sapd/HeadsetControl #238: silent mode

wael444, 2022-10-13:
### Description

for example:
```
$ headsetcontrol -b
Found SteelSeries Arctis (1/7X) Wireless!
Battery: 14%
Success!
```

silent mode should only print out the needed information:
```
$ headsetcontrol -b -s
14%
```
this can be quite useful in use for statusbar scripts.

Sapd, 2022-10-13:
There is an option for that:
`-c` or `--short-output`

wael444, 2022-10-13:
but why did you fucking respond so fast

also that doesnt include neither a newline or the number 'prefix' (%)

Sapd, 2022-10-13:
Some scripts already take use of that option (see https://github.com/Sapd/HeadsetControl#third-party )
So changing that could break those. I also think that i.e. the percent sign is not too important given that you can just add them in your script or using a shell script.

wael444, 2022-10-13:
https://github.com/centic9/headset-charge-indicator/blob/2b63a007b92f27d46708b4d32d391b4203d018f2/headset-charge-indicator.py#L110
well fair, thsnks!

closed 2022-10-13 by wael444