Sapd/HeadsetControl #242: Arctis 9 - detected but not working

w63587, 2022-10-28:
### Description

Hello,
I have just installed HeadsetControl and tried to test it with Arctis 9.
Device is properly detected out of the box, but unfortunately it's not working :/

`~$ headsetcontrol -b
Found SteelSeries Arctis 9!
Failed to set/request battery status, because of timeout, try again.
~$ sudo headsetcontrol -b
Found SteelSeries Arctis 9!
Failed to set/request battery status, because of timeout, try again.
`

Any suggestions, how to fix it?

Best regards

### Headset Name

SteelSeries Arctis 9

### On which OS does the problem happen?

Linux

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
Device Found
  VendorID: 0x1038
 ProductID: 0x12c4
  path: /dev/hidraw8
  serial_number: 000000000000
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 9
  Interface:    5
  Usage-Page: 0x74b7 Usageid: 0x0
 ```

```
Device Found
  VendorID: 0x1038
 ProductID: 0x12c2
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 9
  Interface:    0
  Usage-Page: 0x0 Usageid: 0x0

Device Found
  VendorID: 0x1038
 ProductID: 0x12c2
  path: /dev/hidraw6
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 9
  Interface:    1
  Usage-Page: 0x2e31 Usageid: 0x2f31

Device Found
  VendorID: 0x1038
 ProductID: 0x12c2
  path: /dev/hidraw7
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 9
  Interface:    2
  Usage-Page: 0x0 Usageid: 0x0
```

</details>

mrenvoize, 2022-12-23:
This is now hitting me too on my Arctis 7

rscottyy, 2022-12-30:
https://github.com/Sapd/HeadsetControl/pull/260

try this fix, awaiting tests

fanuch, 2022-12-30:
> try this ... #260 

Legend @rscottyy. 
Tried your branch and it worked!

``` bash

HeadsetControl-s9-fix/build on master 
➜ ./headsetcontrol -b 
Found SteelSeries Arctis 9!
Battery: 62%
Success!
```

closed 2022-12-30 by Sapd