Sapd/HeadsetControl #267: Print headset name when using -c -?

JoePfeiffer, 2023-01-31:
### Description

Thinking about creating a gui, and something I've noticed immediately is that there doesn't seem to be a way to print the headset name when using the -c (--short-output) option.  It would be very helpful if this were available, notably when getting the
headset capabilities.

A suggestion would be to print it on its own line when using -c -? together, so (in the case of my own headset)
% headsetcontrol -c -?
would print
Logitech G533
sbi

This would make it available when setting up the gui.

Sapd, 2023-02-10:
I think it would make more sense to just create a new parameter for that.

Maybe something like `headsetcontrol --print-headset` ?

JoePfeiffer, 2023-02-10:
I like the idea of getting all you need to set it up from a single invocation.

A separate --print-headset would also work (as long as if invoked with -c it only printed the headset name).  It's easy enough to parse anyway that it it's not strictly necessary, but would be nice.

Sapd, 2023-05-01:
See the discussion here: https://github.com/Sapd/HeadsetControl/issues/271

It will probably entirely shift to a chose-able output like json and yaml with more information

closed 2023-05-01 by Sapd