Sapd/HeadsetControl #273: Logitech G535: Battery reports 50%

19Gerhard85, 2023-03-09:
### Description

`$ headsetcontrol -b
Found Logitech G535!
Battery: 50%
Success!
`

### On which OS does the problem happen?

Linux

AXeL-dev, 2023-03-22:
I'm facing the same issue, any news?

[image]

AXeL-dev, 2023-03-22:
@19Gerhard85 I was able to fix the issue by reverting the changes in this commit: https://github.com/Sapd/HeadsetControl/commit/5a09dc3401afb344c4e7dc9b6bff83708426ef40.

Sapd, 2023-03-22:
> @19Gerhard85 I was able to fix the issue by reverting the changes in this commit: [5a09dc3](https://github.com/Sapd/HeadsetControl/commit/5a09dc3401afb344c4e7dc9b6bff83708426ef40).

Best way is to look up why this happens. I would suggest you adding this around 170:

```
printf("voltage %u - result: %d\n", voltage, spline_battery_level(battery_estimate_percentages, battery_estimate_voltages, battery_estimate_size, voltage));
```

AXeL-dev, 2023-03-22:
Hum :thinking: , i don't think that there is a bug there, but the current behavior only estimates the battery level based on the `battery_estimate_percentages`  variable:

https://github.com/Sapd/HeadsetControl/blob/5a09dc3401afb344c4e7dc9b6bff83708426ef40/src/devices/logitech_g535.c#L61

So only values referenced in that variable would be used to display the battery level, example: if your battery level is below `100%`  the returned value would always be `50%` until your battery level reaches a value below that, like `49%` & then the displayed value would be `30%` & not `49%` since only referenced values in the `battery_estimate_percentages`variable would be used. That's the theory that seems more logical for me.

PR #291 (2023-05-01): Fix spline function
Resolves #273
commit afcaa04a (2023-05-01): Fix spline function (#291)

Resolves #273
closed 2023-05-01 by Sapd