Sapd/HeadsetControl #274: Add Audeze Maxwell support for Linux

MurzNN, 2023-03-16:
### Description

Will be good to add Audeze Maxwell support for HeadsetControl, at least turning on/off sidetone action, and battery state.

### Headset Name

Audeze Maxwell

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
 PASTE HERE
 ```

</details>

Armadill0, 2024-01-23:
Yes please! Unstale.

inorichi, 2024-05-26:
I found the battery state using Wireshark, however the value doesn't update automatically. The Audeze app sends several SET_REPORT events, which probably are the ones that update the value, but I was not able to replicate them (it even started sending wrong values on the GET_REPORT).

I'll attach the Wireshark log in case anyone with better USB notions than me can work it out. I opened the app while the headset was charging and it went from 97% to 99% then I stopped it.

In frame 2008 the hex at 0028 changed from 0x61 to 0x62 (97% to 98%).
In frame 4300 the hex at 0028 changed from 0x62 to 0x63 (98% to 99%).

[audeze 97 to 99.pcapng.txt](https://github.com/Sapd/HeadsetControl/files/15446816/audeze.97.to.99.pcapng.txt) (remove the .txt extension)

Xrey274, 2024-08-14:
I'd be interested in controlling the turn off time, so that'd be great.

ZlordHUN, 2024-08-25:
+1. Would be nice to add Maxwell support.

Sapd, 2025-04-04:
I see there is high demand :) But note that only people who actually own the headset can implement that. Here is a small tutorial: https://github.com/Sapd/HeadsetControl/wiki/Development-1-%E2%80%90-Adding-a-device

ModoloDev, 2025-05-17:
I’ve opened a PR adding Battery, Sidetone, Inactive Time, Volume Limiter, and Equalizer Preset capabilities for Audeze Maxwell #401

MurzNN, 2025-05-22:
Yeah, good job @ModoloDev - thank you a lot!

closed 2025-05-22 by MurzNN