Sapd/HeadsetControl #275: Add support for Jabra Evolve2 series (65, 75, 85)

MurzNN, 2023-03-17:
### Description

Will be good to add support for Jabra Evolve2 devices series (65, 75, 85).

### Headset Name

Jabra Evolve2 65

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
 PASTE HERE
 ```

</details>

DaSilva2010, 2024-01-22:
Are there any plans to implement this? I would appreciate.

deraffe, 2024-11-18:
Being able to control the busy LED on the [Evolve2 85](https://www.jabra.com/business/office-headsets/jabra-evolve/jabra-evolve2-85) would be nice. It looks like the only way to do that under Linux is currently using the [Jabra SDK](https://developer.jabra.com/site/global/sdk/index.gsp) and write your own application?