Sapd/HeadsetControl #277: Steelseries Nova Pro Wireless (New revision)

Eb-, 2023-03-19:
### Description

Picked up a Nova Pro Wireless, unfortunately it looks like it's a new revision that isn't recognized by `headsetcontrol`. Looks like this one has a ProductID of 0x12e5 instead of the expected 0x12e0.

### Headset Name

Steelseries Nova Pro Wireless

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
 $ headsetcontrol --dev -- --list --device 0x1038:0x12e5
Device Found
  VendorID: 0x1038
 ProductID: 0x12e5
  path: /dev/hidraw16
  serial_number: 
  Manufacturer: SteelSeries
  Product:      Arctis Nova Pro Wireless
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12e5
  path: /dev/hidraw17
  serial_number: 
  Manufacturer: SteelSeries
  Product:      Arctis Nova Pro Wireless
  Interface:    4
  Usage-Page: 0xffc0 Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12e5
  path: /dev/hidraw17
  serial_number: 
  Manufacturer: SteelSeries
  Product:      Arctis Nova Pro Wireless
  Interface:    4
  Usage-Page: 0xff00 Usageid: 0x1
 ```

</details>

Eb-, 2023-03-19:
For reference a request for a different revision of the Nova Pro Wireless is found here: https://github.com/Sapd/HeadsetControl/issues/221

Sapd, 2023-03-20:
Best thing is to try to add the id yourself and then make a PR. As only people with this headset are able to test it.

You can do so according to the Wiki of how to add a Corsair device, but instead you simply use the file which belongs to your headset. Otherwise the steps are similar: https://github.com/Sapd/HeadsetControl/wiki/Adding-a-Corsair-device