Sapd/HeadsetControl #281: HID error on macOS 13.3

oscarsommerer, 2023-03-31:
### Description

Ever since updating to macOS Ventura 13.3 a few days ago `headsetcontrol` has stopped working for me. I'm getting some very generic sound error messages, eg:
```
$ headsetcontrol -b
Found Corsair Headset Device!
Requested/supported HID device not found or system error.
 HID Error: Success
```

### Headset Name

Corsair VOID PRO Wireless

### On which OS does the problem happen?

MacOS

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a1a
  path: DevSrvsID:4294980423
  serial_number:
  Manufacturer: Corsair
  Product:      Corsair VOID PRO Wireless Gaming Headset
  Interface:    0
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a1a
  path: DevSrvsID:4294980423
  serial_number:
  Manufacturer: Corsair
  Product:      Corsair VOID PRO Wireless Gaming Headset
  Interface:    0
  Usage-Page: 0xffc5 Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a1a
  path: DevSrvsID:4294980423
  serial_number:
  Manufacturer: Corsair
  Product:      Corsair VOID PRO Wireless Gaming Headset
  Interface:    0
  Usage-Page: 0xff00 Usageid: 0x1
 ```

</details>

Sapd, 2023-03-31:
I verified it, and its a bit weird. Seems that with the new Mac update the interface id changed (to 0), while on a Linux VM it is still 3. So I simply changed it to 0 in the latest master which means that simply the first interface will be used.

oscarsommerer, 2023-03-31:
Ha! Interesting development.

I've installed the latest version and can confirm it started working again. Thanks for the quick fix! 😄

elaw-fastly, 2023-04-26:
Same issue applies to SteelSeries Arctis 7 on macOS 13.3.1
changing .interface = 0x05 to 0x00 fixed it.

Sapd, 2023-04-27:
> Same issue applies to SteelSeries Arctis 7 on macOS 13.3.1 changing .interface = 0x05 to 0x00 fixed it.
> 
> I found someone referencing the issue here: https://developer.apple.com/forums/thread/728001

That could be indeed an Apple bug. Maybe it also makes sense to create an issue at the libusb/hidapi repo, so that they also can check.

In theory we can also simply set interfaceid to 0 for all devices which only use one.

Sapd, 2023-04-30:
I thought about it again, and I think the best way to fix it, is to change this function: https://github.com/Sapd/HeadsetControl/blob/664122495b9bb38a8b86b89fb9f2133a620eb0f7/src/hid_utility.c#L24

and set iid to 0 for macos. Will reopen the issue until it is done

closed 2023-05-01 by Sapd

Sapd, 2024-05-19:
WIll keep the code-as is, but for further reference should an issue arise:

Fixed here in HIDAPI: https://github.com/libusb/hidapi/pull/534
Seems like an undocumented property was used, and Apple removed that during an update: https://github.com/libusb/hidapi/issues/531

commit 40d2d008 (2023-03-31): Corsair Void: Use first interface

Mac seems to user a different
interfaceid now
We simply take now the first
enumerated interfaceid (0)

Fixes #281