Sapd/HeadsetControl #285: Generating udev rules for Artics Nova 7

anyoneyun, 2023-04-15:
Hi, I've been trying to use HeadsetControl with a pair of Artics Nova 7, it works fine when using sudo, but running it as a normal user doesn't work, even after generating udev rules with `sudo headsetcontrol -u`. In /usr/lib/udev/rules.d/70-headsets.rules there is no Artics Nova 7. Is it intended? Thanks!

Sapd, 2023-04-16:
Did you write the generated ids down? i.e.
`sudo headsetcontrol -u > /usr/lib/udev/rules.d/70-headsets.rules`

If the headset works with sudo, the id actually must also be inside the rules (as the udev rules generation just iterates all headsets)

anyoneyun, 2023-04-16:
Hi, I've tried doing that and it does actually show ids for Artics Nova 7 now. However, I had to move the 70-headsets.rules to /etc/udev/rules.d to make it work. Thanks!

closed 2023-04-16 by anyoneyun