Sapd/HeadsetControl #294: SteelSeries Arctis Nova 3 support

socketpair, 2023-05-03:
### Description

Bus 001 Device 014: ID 1038:12ec SteelSeries ApS Arctis Nova 3

### Headset Name

SteelSeries Arctis Nova 3

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
Device Found
  VendorID: 0x1038
 ProductID: 0x12ec
  path: /dev/hidraw3
  serial_number: 
  Manufacturer: SteelSeries
  Product:      Arctis Nova 3
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12ec
  path: /dev/hidraw4
  serial_number: 
  Manufacturer: SteelSeries
  Product:      Arctis Nova 3
  Interface:    4
  Usage-Page: 0xffc0 Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12ec
  path: /dev/hidraw4
  serial_number: 
  Manufacturer: SteelSeries
  Product:      Arctis Nova 3
  Interface:    4
  Usage-Page: 0xff00 Usageid: 0x1

 ```

</details>

AnErrupTion, 2023-12-27:
@socketpair @Sapd I also have this headset and wanted to add support for it in HeadsetControl. However, I'm hitting a slightly problematic roadblock here... When I end up sending a feature report, I'm getting this error:

``Failed to set/request sidetone. Error: -1: ioctl (SFEATURE): Broken pipe``

I also tried using PyUSB to send the feature report instead, but then I'm getting first a "Resource busy" error, then "Entity not found" (when re-running the program). I'm not exactly sure what's wrong, the headset is working perfectly fine.

Here's my code for setting the sidetone:
```c
static int arctis_nova_3_send_sidetone(hid_device* device_handle, uint8_t num)
{
    /*
     * steps to make it work nicely with headset-charge-indicator
     */
    if (num < 26) {
        num = 0x0;
    } else if (num < 51) {
        num = 0x1;
    } else if (num < 76) {
        num = 0x2;
    } else {
        num = 0x3;
    }

    uint8_t data[MSG_SIZE + 1] = { 0x02, 0x06, 0x39, num };

    return hid_send_feature_report(device_handle, data, sizeof(data));
}
```

I've also attached the Wireshark capture file below (inside a ZIP file) which shows the packets when I set the microphone sidetone to "Low" in SteelSeries GG Engine (which corresponds to the value ``0x01``):
[ArctisNova3.zip](https://github.com/Sapd/HeadsetControl/files/13774775/ArctisNova3.zip)

AnErrupTion, 2023-12-27:
Nevermind, I guess I shouldn't code at 3 AM... I solved it by just using this as the data buffer:
``uint8_t data[MSG_SIZE] = { 0x06, 0x39, num };``
And now setting the microphone sidetone works!

PR #319 (2023-12-27): Add Arctis Nova 3 support
Hello! This PR adds support for the Arctis Nova 3 headset. Alongside this, it adds 2 new capabilities that this headset implements, those being the following:

- ``CAP_MICROPHONE_MUTE_LED_BRIGHTNESS``
- ``CAP_MICROPHONE_VOLUME``

The headset also implements the following existing capabilities:

- ``CAP_SIDETONE``
- ``CAP_EQUALIZER_PRESET``
- ``CAP_EQUALIZER``

Closes #294.
closed 2023-12-27 by Sapd