Sapd/HeadsetControl #297: Logitech G733 not recognised

afettouhi, 2023-05-21:
I just got a Logitech G733 headset that I am using under Arch Linux. I installed headsetcontrol and headsetkontrol from AUR. But headsetcontrol can't seem to find it. If I run 

headsetcontrol --dev -- --list

I get

Device Found
  VendorID: 0x046d
 ProductID: 0x0afe
  path: /dev/hidraw5
  serial_number: 0000000000000000
  Manufacturer: Logitech
  Product:      G733 Gaming Headset
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x046d
 ProductID: 0x0afe
  path: /dev/hidraw5
  serial_number: 0000000000000000
  Manufacturer: Logitech
  Product:      G733 Gaming Headset
  Interface:    3
  Usage-Page: 0xff43 Usageid: 0x202

Device Found
  VendorID: 0x046d
 ProductID: 0x0afe
  path: /dev/hidraw5
  serial_number: 0000000000000000
  Manufacturer: Logitech
  Product:      G733 Gaming Headset
  Interface:    3
  Usage-Page: 0xff13 Usageid: 0x1

but I can't connect to it neither as a user or as root.

new69, 2023-05-26:
Same problem here, with same device

Sapd, 2023-05-27:
What is the error message? Are you using the latest version of headsetcontrol from git?

afettouhi, 2023-05-27:
I am using the latest stable version from AUR on Arch Linux. I don't see any error messages at all.

Sapd, 2023-05-27:
That is not the latest, as far as I see there is a headsetcontrol-git package, you can either try this or try to compile from master

afettouhi, 2023-05-27:
Ok it works with the git branch. I can now see my headset and see the battery status. All good, many thanks!

closed 2023-05-27 by Sapd