Sapd/HeadsetControl #299: Can't seem to install

GoodMew, 2023-05-28:
### Description

Following your Ubuntu instructions, I get up to the 'make' command in the build directory, and get:
make: /usr/bin/cmake: No such file or directory
make: *** [Makefile:1307: cmake_check_build_system] Error 127

I have tried removing everything and starting over, but same result. I'm not sure where else to get help. It worked fine on my PopOS distro!

### On which OS does the problem happen?

Linux

Sapd, 2023-05-29:
Did the cmake .. command work before without errors ?
What is the output of `which cmake`?

GoodMew, 2023-06-11:
Sorry for the late reply, I kinda put maining Linux on the backburner after running into this issue. 
So cmake .. results in:
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found hidapi: /usr/lib/x86_64-linux-gnu/libhidapi-hidraw.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/spyvos/HeadsetControl/HeadsetControl/build

I don't think I've used the cmake .. command before this, but that seems alright. Then the make command results in the same Error 127. The output of which cmake is simply:
/home/spyvos/.local/bin/cmake

I'm sorry to be a fool, surely there's a simple fix.. But your help is greatly appreciated, I desperately want to stop using Windows, however my headset working well is critical for my income.

GoodMew, 2023-06-11:
I managed to get it working!! I used pkexec abroot shell command and then just followed your PopOS instructions from the start, it works great. TYSM! <3