Sapd/HeadsetControl #302: Extend the project to others devices

lolio-dev, 2023-06-20:
### Description

Hi, I use a steelseries aerox 3 mouse Wireless and I would like to get the battery percentage to display it on a custom streamdeck, this project looks perfect but only for headsets... Are there any plans to extend the project to mouse and others wireless device ? 
It can also be interesting to explain in the README how to retrieve the data because it's not an easy thing to find how to do it.

Thank,
Elie

Sapd, 2023-06-20:
In principle, there is nothing to be said against this. It also makes no real difference for the software whether it is a headset or a mouse. So I would also accept PRs with other (HID) devices.

> It can also be interesting to explain in the README how to retrieve the data because it's not an easy thing to find how to do it.

In the readme there is actually a link to the wiki under the Development section which explains it: https://github.com/Sapd/HeadsetControl/wiki/Development

Battery is a little bit more difficult than Sidetone, however the steps are similar. Basically you need to search for a decreasing value in a packet which is to the **host**.
Often the windows driver sends a command first to tell the headset, that it should send battery data, which you can also find out afterwards.

I would advise completely reading the wiki article before trying it. If you struggle you can always come back here and report your progress or ask.

lolio-dev, 2023-06-20:
Okay, thank you, I'll think about it.

Sapd, 2024-04-16:
This is now easier to do. The last update includes ground-work to also support multiple devices (not fully implemented in code, but not much is missing)

closed 2025-04-12 by github-actions[bot]