Sapd/HeadsetControl #309: Should \n be added to short output?

Odizinne, 2023-10-28:
### Description

Hello, not sure if this is wanted or not but:

```
flora@pop-os:~$ headsetcontrol -b -c
-2flora@pop-os:~$ 
```
This is what i get when shortening output.
Is this something intended?

I would expect something like this:

```
flora@pop-os:~$ headsetcontrol -b -c
-2
flora@pop-os:~$ 
```

### On which OS does the problem happen?

Linux

Sapd, 2023-10-29:
It probably was done this way to make it easier to directly parse. But I agree with you, normally a Linux command should always end with \n.

However, I wouldn't change it due to backwards compatibility. Also because `-c` will be probably deprecated in the future in favour of https://github.com/Sapd/HeadsetControl/issues/271 (Feel free to participate in the discussion)

Sapd, 2024-04-12:
Closed because short output is now deprecated in favour of JSON/YAML output

closed 2024-04-12 by Sapd