Sapd/HeadsetControl #314: Support for Logitech G Pro X 2

darkymtp, 2023-12-20:
### Description

Hello.
I can set sidetone, noise reduction and timeout via these command lines : 
Side tone :
```
headsetcontrol --dev -- --device 0x046d:0x0af7 --usage 0xffa0,0x1 --send "0x51, 0x0a, 0x00, 0x03, 0x1b, 0x00, 0x05, 0x00, 0x07, 0x1b, 0x01, $1"
```
Mic noise reduction :
```
headsetcontrol --dev -- --device 0x046d:0x0af7 --usage 0xffa0,0x1 --send "0x51, 0x09, 0x00, 0x03, 0x1c, 0x00, 0x04, 0x00, 0x08, 0x1c, $1"
```
Time out : 
```
headsetcontrol --dev -- --device 0x046d:0x0af7 --usage 0xffa0,0x1 --send "0x51, 0x09, 0x00, 0x03, 0x1c, 0x00, 0x04, 0x00, 0x06, 0x1d, $1"
```
I forked the project and wrote files to support side tone settings only, unsuccessfully.
https://github.com/darkymtp/HeadsetControl/

Help ! :-)

### Headset Name

Logitech G Pro X 2

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
 Device Found
  VendorID: 0x046d
 ProductID: 0x0af7
  path: /dev/hidraw4
  serial_number: 0000000000000000
  Manufacturer: Logitech
  Product:      PRO X 2 LIGHTSPEED
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x046d
 ProductID: 0x0af7
  path: /dev/hidraw4
  serial_number: 0000000000000000
  Manufacturer: Logitech
  Product:      PRO X 2 LIGHTSPEED
  Interface:    3
  Usage-Page: 0xff13 Usageid: 0x1

Device Found
  VendorID: 0x046d
 ProductID: 0x0af7
  path: /dev/hidraw4
  serial_number: 0000000000000000
  Manufacturer: Logitech
  Product:      PRO X 2 LIGHTSPEED
  Interface:    3
  Usage-Page: 0xffa0 Usageid: 0x1
 ```

</details>

Sapd, 2023-12-20:
Hello,

> Hello. I can set sidetone, noise reduction and timeout via these command lines : Side tone :
> 
> ```
> headsetcontrol --dev -- --device 0x046d:0x0af7 --usage 0xffa0,0x1 --send "0x51, 0x0a, 0x00, 0x03, 0x1b, 0x00, 0x05, 0x00, 0x07, 0x1b, 0x01, $1"
> ```

https://github.com/darkymtp/HeadsetControl/blob/1e0255409ee73cccce32bb75a358cd9e7891f4b1/src/devices/logitech_gpro_x2.c#L40

If that works, your command here must be different. HIDPP_LONG_MESSAGE is just a value for `0x11`, which is not strictly applicable to all (logitech) headsets.

So the code must simply look like this:
```
 uint8_t sidetone_data[12] = { 0x51, 0x0a, 0x00, 0x03, 0x1b, 0x00, 0x05, 0x00, 0x07, 0x1b, 0x01, num};
```

also dont forget to set usage and usagepage in gpro_x2_init

darkymtp, 2023-12-20:
It works, thanks !

Sapd, 2023-12-21:
Good to hear, dont forget to PR it! (Just click on Pull Requests, and create a new one from the branch where you pushed the changes)

darkymtp, 2023-12-21:
I would like to add battery status before PR.

darkymtp, 2023-12-21:
I have no clue on how to add battery status reporting, so here is my contribution.

closed 2023-12-24 by darkymtp