Sapd/HeadsetControl #317: Is it possible to get your default sidetone level from somewhere?

RadoslavL, 2023-12-25:
I know I am kinda going out of bounds of this github repository a bit, but I don't really know where else to find information about its sidetone level format. I already saw #313, and I am aware of the fact that the program is unable to retrieve information from the headset. I was just wondering if it is possible to get this information from somewhere. I have a G432 headset and I really don't want to lose the default setting. Is there a website, where this kind of information about the headset is listed? I am struggling to understand the format as a whole.

Sapd, 2023-12-25:
> I have a G432 headset and I really don't want to lose the default setting

Usually the headsets (there could be exceptions but I did not see any) don't save the settings permanently. That means if you plug it out/in and restart the headset all the default settings should be back again.

RadoslavL, 2023-12-25:
I guess mine saves its settings. I disconnected and reconnected it a couple of times and it kept the sidetone volume.

closed 2023-12-25 by RadoslavL