Sapd/HeadsetControl #321: Corsair HS80 MAX WIRELESS support

Rexarn, 2023-12-28:
### Description

Please add this one, i don't wanna download IQue.
I do not know which device id is the "real" one so i pasted in everything it gave me.

### Headset Name

Corsair HS80 MAX WIRELESS

### Device information

<details>
 <summary>Detailed Device Information</summary>

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a96
  path: \\?\HID#VID_1B1C&PID_0A96&Col01#7&246a0ded&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 7BD8BF8ABF5ED065
  Manufacturer: Corsair
  Product:      CORSAIR HS80 MAX WIRELESS Gaming Headset
  Interface:    -1
  Usage-Page: 0xff13 Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a96
  path: \\?\HID#VID_1B1C&PID_0A96&Col02#7&246a0ded&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 7BD8BF8ABF5ED065
  Manufacturer: Corsair
  Product:      CORSAIR HS80 MAX WIRELESS Gaming Headset
  Interface:    -1
  Usage-Page: 0xb Usageid: 0x5

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a96
  path: \\?\HID#VID_1B1C&PID_0A96&Col03#7&246a0ded&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 7BD8BF8ABF5ED065
  Manufacturer: Corsair
  Product:      CORSAIR HS80 MAX WIRELESS Gaming Headset
  Interface:    -1
  Usage-Page: 0xff42 Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a96
  path: \\?\HID#VID_1B1C&PID_0A96&Col04#7&246a0ded&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: 7BD8BF8ABF5ED065
  Manufacturer: Corsair
  Product:      CORSAIR HS80 MAX WIRELESS Gaming Headset
  Interface:    -1
  Usage-Page: 0xff42 Usageid: 0x2

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a97
  path: \\?\HID#VID_1B1C&PID_0A97&MI_03&Col01#7&2c0f6295&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: B8CE92D9ABAE85DF
  Manufacturer: Corsair
  Product:      CORSAIR HS80 MAX WIRELESS Gaming Receiver
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a97
  path: \\?\HID#VID_1B1C&PID_0A97&MI_03&Col02#7&2c0f6295&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: B8CE92D9ABAE85DF
  Manufacturer: Corsair
  Product:      CORSAIR HS80 MAX WIRELESS Gaming Receiver
  Interface:    3
  Usage-Page: 0xff13 Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a97
  path: \\?\HID#VID_1B1C&PID_0A97&MI_03&Col03#7&2c0f6295&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: B8CE92D9ABAE85DF
  Manufacturer: Corsair
  Product:      CORSAIR HS80 MAX WIRELESS Gaming Receiver
  Interface:    3
  Usage-Page: 0xb Usageid: 0x5

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a97
  path: \\?\HID#VID_1B1C&PID_0A97&MI_03&Col04#7&2c0f6295&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: B8CE92D9ABAE85DF
  Manufacturer: Corsair
  Product:      CORSAIR HS80 MAX WIRELESS Gaming Receiver
  Interface:    3
  Usage-Page: 0xff42 Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a97
  path: \\?\HID#VID_1B1C&PID_0A97&MI_03&Col05#7&2c0f6295&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: B8CE92D9ABAE85DF
  Manufacturer: Corsair
  Product:      CORSAIR HS80 MAX WIRELESS Gaming Receiver
  Interface:    3
  Usage-Page: 0xff42 Usageid: 0x2

</details>

closed 2024-12-23 by github-actions[bot]

nafg, 2026-01-13:
Claude says HS80 MAX is different than HS80 and it couldn't get those solutions to work

Sapd, 2026-01-13:
No still fine, provides some insights anyways and also summarizes the problem clearly