Sapd/HeadsetControl #322: Add option to check if headset is connected

AinoSpring, 2024-01-27:
### Description

When writing scripts using headsetcontrol, it would be very practical having
a flag which checks if there is a headset connected.

Sapd, 2024-01-27:
Will be provided indirectly over 
https://github.com/Sapd/HeadsetControl/issues/271
(Feel free to comment there for feedback)

AinoSpring, 2024-01-28:
Yeah but when writing stuff like bash scripts (e. g. for an i3 status bar module) it
would be really convenient if you could do something like

`if [ headsetcontrol --connected ] ; then`

and it feels like it should exist.

Sapd, 2024-01-28:
Makes sense, will probably implement that and --count or so to check how many are connected (used for further use cases).

closed 2024-02-01 by Sapd