Sapd/HeadsetControl #325: Failed to set/request sidetone. Error: -1: ioctl (SFEATURE): Broken pipe

TigerTy9, 2024-02-01:
### Description

0x0a53 
 
Error when trying to adjust sidetone.

"Failed to set/request sidetone. Error: -1: ioctl (SFEATURE): Broken pipe"

### Headset Name

Corsair VOID ELITE Surround USB Adapter

### On which OS does the problem happen?

Linux

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
 PASTE HERE
 ```
Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a53
  path: /dev/hidraw8
  serial_number: 00000000
  Manufacturer: Cosair
  Product:      Corsair VOID ELITE Surround USB Adapter
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0xc9
</details>

Sapd, 2024-02-01:
Im a bit confused, because here you are saying it works: https://github.com/Sapd/HeadsetControl/pull/324

stuarthayhurst, 2024-07-29:
Implemented this in my driver (https://github.com/stuarthayhurst/corsair-void-driver/compare/53ef75de686152aed86be49ebc22bd5120bb0ac6...f74639e207a83d61c557aeeecf7a7b6a7bf3b87c), so it should work here too

closed 2025-07-25 by github-actions[bot]

hidminds, 2025-10-06:
I have the same issue with CachyOS Linux tars 6.17.0-4-cachyos #1 SMP PREEMPT_DYNAMIC Fri, 03 Oct 2025 17:06:47 +0000 x86_64 GNU/Linux

headsetcontrol -s 64
```
Found 1 supported device(s):
 Corsair Headset Device (Corsair VOID ELITE USB Gaming Headset) [0x1b1c:0x0a52]

Failed to set/request sidetone. Error: -1: ioctl (SFEATURE): Broken pipe
```

lsusb Output:
`Bus 001 Device 019: ID 1b1c:0a52 Corsair Corsair VOID ELITE USB Gaming Headset`

modinfo hid_corsair_void
```
filename:       /lib/modules/6.17.0-4-cachyos/kernel/drivers/hid/hid-corsair-void.ko.zst
license:        GPL
author:         Stuart Hayhurst <stuart.a.hayhurst@gmail.com>
description:    HID driver for Corsair Void headsets
name:           hid_corsair_void
intree:         Y
depends:        
alias:          hid:b0003g*v00001B1Cp00000A0C
alias:          hid:b0003g*v00001B1Cp00000A2B
alias:          hid:b0003g*v00001B1Cp00001B23
alias:          hid:b0003g*v00001B1Cp00001B25
alias:          hid:b0003g*v00001B1Cp00001B27
alias:          hid:b0003g*v00001B1Cp00000A0F
alias:          hid:b0003g*v00001B1Cp00001B1C
alias:          hid:b0003g*v00001B1Cp00001B29
alias:          hid:b0003g*v00001B1Cp00001B2A
alias:          hid:b0003g*v00001B1Cp00000A30
alias:          hid:b0003g*v00001B1Cp00000A31
alias:          hid:b0003g*v00001B1Cp00000A14
alias:          hid:b0003g*v00001B1Cp00000A16
alias:          hid:b0003g*v00001B1Cp00000A1A
alias:          hid:b0003g*v00001B1Cp00000A17
alias:          hid:b0003g*v00001B1Cp00000A1D
alias:          hid:b0003g*v00001B1Cp00000A18
alias:          hid:b0003g*v00001B1Cp00000A1E
alias:          hid:b0003g*v00001B1Cp00000A1F
alias:          hid:b0003g*v00001B1Cp00000A51
alias:          hid:b0003g*v00001B1Cp00000A55
alias:          hid:b0003g*v00001B1Cp00000A75
alias:          hid:b0003g*v00001B1Cp00000A52
alias:          hid:b0003g*v00001B1Cp00000A56
alias:          hid:b0003g*v00001B1Cp00000A53
alias:          hid:b0003g*v00001B1Cp00000A57
srcversion:     A790344FF24F988CB1BFF26
vermagic:       6.17.0-4-cachyos SMP preempt mod_unload 
retpoline:      Y
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        16:37:75:CC:AF:DC:FF:D7:37:08:15:68:E0:EC:5F:78:8E:36:2C:1B
sig_hashalgo:   sha512
signature:      30:65:02:31:00:97:47:3C:9E:7E:E8:28:52:1B:DB:51:74:F5:F4:39:
                C1:7D:43:C7:0E:56:46:2D:A1:4D:AB:C7:9D:88:80:19:63:B6:4C:60:
                68:C9:B9:04:CA:9B:AB:A3:C7:5B:1B:32:3E:02:30:60:96:82:5C:C1:
                8C:7F:20:4E:1D:34:92:40:07:CB:6A:FC:49:88:1F:34:DE:72:AD:ED:
                66:90:D1:7C:B5:EF:E3:0D:F7:CC:8A:07:66:B7:E8:12:A2:5C:A9:68:
                88:F0:05

```

paru -Q headsetcontrol
`headsetcontrol 3.1.0-2.1`

headsetcontrol -o ENV
```
HEADSETCONTROL_NAME="HeadsetControl"
HEADSETCONTROL_VERSION=""
HEADSETCONTROL_API_VERSION="1.3"
HEADSETCONTROL_HIDAPI_VERSION="0.15.0"
DEVICE_COUNT=1
DEVICE_0="Corsair Headset Device"
DEVICE_0_CAPABILITIES_AMOUNT=4
DEVICE_0_CAPABILITY_0="CAP_SIDETONE"
DEVICE_0_CAPABILITY_1="CAP_BATTERY_STATUS"
DEVICE_0_CAPABILITY_2="CAP_NOTIFICATION_SOUND"
DEVICE_0_CAPABILITY_3="CAP_LIGHTS"
DEVICE_0_BATTERY_STATUS="BATTERY_UNAVAILABLE"
DEVICE_0_BATTERY_LEVEL=-1
DEVICE_0_ERROR_COUNT=0

```