Sapd/HeadsetControl #330: SteelSeries Arctis 9  not displaying battery when charging

Novattz, 2024-03-03:
### Description

As the title explains when my wireless Steelseries Arctis 9's are charging and i do headsetcontrol -b the output just displays "charging" and nothing else. i would like to get the battery % while also charging. I do not know if this is intended or just a limitation based on the headset but i would atleast like to get confirmation if this is intended or not and if its fixable in any other way on my end.

### Headset Name

SteelSeries Arctis 9

### On which OS does the problem happen?

Linux

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
 Device Found
  VendorID: 0x1038
 ProductID: 0x12c2
  path: /dev/hidraw9
  serial_number:
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 9
  Interface:    0
  Usage-Page: 0xffc0 Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12c2
  path: /dev/hidraw10
  serial_number:
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 9
  Interface:    1
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12c2
  path: /dev/hidraw12
  serial_number:
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis 9
  Interface:    2
  Usage-Page: 0xff00 Usageid: 0x1
 ```

</details>

Sapd, 2024-03-03:
Its intended. There is a flag inside the battery variable which is set when its charging.
 Its probably possible to get the level anyway, but it will most likely deviate/jump extremely (for example you would plug it in, charge will instantly go to 20%, you plug it out it would plumbed). 
Thats because (in comparison to for example smartphones), battery value is often just derived from voltage, which is highly inaccurate. Smartphones count it by measuring the amount of energy (ampers cumulated) went through.

Novattz, 2024-03-03:
> Its intended. There is a flag inside the battery variable which is set when its charging. Its probably possible to get the level anyway, but it will most likely deviate/jump extremely (for example [...]

since u mentioned that its possible is it something that could be added? i dont really care if its inaccurate i just want a general percentage so it doesnt just say "charging"

Sapd, 2024-03-07:
> since u mentioned that its possible is it something that could be added? i dont really care if its inaccurate i just want a general percentage so it doesnt just say "charging"

Possibly yes. You will have to take a look here: https://github.com/Sapd/HeadsetControl/blob/a95a015b7aa094f5369c861ded094ecb7eb3a45b/src/devices/steelseries_arctis_9.c#L95

Problem is a bit, that the function currently can only return one thing (either the level or a status information). I can change that in the next version

Sapd, 2024-03-14:
When looking at it again, you can simply "fix it" for now, by just removing 

```
    if (data_read[4] == 0x01)
        return BATTERY_CHARGING;
 ```

I first thought that it is saved as flag, but it is saved in a different byte so just removing that two lines should be sufficient.

Sapd, 2024-04-01:
Released in latest version

closed 2024-04-01 by Sapd