Sapd/HeadsetControl #335: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]

Leixb, 2024-04-03:
### Description

When trying to build the project from source, it fails with the following error:

```yaml
error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
```

This seems to be caused because [glibc contains its own definition of `asprintf`](https://codebrowser.dev/glibc/glibc/libio/stdio.h.html#396) which has the `warn_unused_result` attribute.

If I remove `-Werror` and build the binary, everything works as expected and the `asprintf` linked
in the final binary is the one from your `utility.h`, so it's quite strange that `gcc` complains about it.

```console
$ objdump -T headsetcontrol | grep asprint
0000000000407650 g    DF .text	00000000000000b8  Base        asprintf
0000000000407510 g    DF .text	000000000000013c  Base        vasprintf
```

For completeness, here is the full build log:
```console
$ nix-shell -p cmake hidapi
$ mkdir build && cd build
$ cmake ..
$ make
[  3%] Building C object CMakeFiles/headsetcontrol.dir/src/main.c.o
/home/leix/Documents/github/HeadsetControl/src/main.c: In function ‘handle_feature’:
/home/leix/Documents/github/HeadsetControl/src/main.c:197:9: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
  197 |         asprintf(&result.message, "This headset doesn't support %s", capabilities_str[cap]);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leix/Documents/github/HeadsetControl/src/main.c:208:13: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
  208 |             asprintf(&result.message, "Could not open device. Error: %ls", hid_error(*device_handle));
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leix/Documents/github/HeadsetControl/src/main.c:229:13: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
  229 |             asprintf(&result.message, "Battery: %d%%", battery.level);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leix/Documents/github/HeadsetControl/src/main.c:247:17: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
  247 |                 asprintf(&result.message, "Error retrieving battery status. Error: %ls", hid_error(*device_handle));
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leix/Documents/github/HeadsetControl/src/main.c:249:17: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
  249 |                 asprintf(&result.message, "Error retrieving battery status");
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leix/Documents/github/HeadsetControl/src/main.c:272:13: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
  272 |             asprintf(&result.message, "Chat-Mix: %d", ret);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leix/Documents/github/HeadsetControl/src/main.c:327:9: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
  327 |         asprintf(&result.message, "Failed to set/request %s, because of timeout", capabilities_str[cap]);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leix/Documents/github/HeadsetControl/src/main.c:330:9: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
  330 |         asprintf(&result.message, "Failed to set/request %s. HeadsetControl Error", capabilities_str[cap]);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leix/Documents/github/HeadsetControl/src/main.c:333:9: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
  333 |         asprintf(&result.message, "Failed to set/request %s. Provided parameter out of boundaries", capabilities_str[cap]);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leix/Documents/github/HeadsetControl/src/main.c:337:13: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
  337 |             asprintf(&result.message, "Failed to set/request %s. Error: %d: %ls", capabilities_str[cap], ret, hid_error(*device_handle));
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/leix/Documents/github/HeadsetControl/src/main.c:339:13: error: ignoring return value of ‘asprintf’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
  339 |             asprintf(&result.message, "Failed to set/request %s. Error: %d", capabilities_str[cap], ret);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/headsetcontrol.dir/build.make:76: CMakeFiles/headsetcontrol.dir/src/main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:149: CMakeFiles/headsetcontrol.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

Here are some environment details:

```console
$ uname -a
Linux kuro 6.7.11 #1-NixOS SMP PREEMPT_DYNAMIC Tue Mar 26 22:22:50 UTC 2024 x86_64 GNU/Linux

$ cmake --version
cmake version 3.28.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).

$ make --version
GNU Make 4.4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

$ gcc --version
gcc (GCC) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

### On which OS does the problem happen?

Linux

TheBill2001, 2024-04-03:
I'm currently the maintainer of the [AUR package](https://aur.archlinux.org/packages/headsetcontrol) and can confirm this issue. Arch Linux's build system contains a lot of additional compiler flags, which triggers a bunch of warning. And with `-Werror`, the build just fails. The current workaround is to omit the following line in the CMake file:

https://github.com/Sapd/HeadsetControl/blob/0005a0a5fd9c27b02d8fb49f138a34cd2ddd4c78/CMakeLists.txt#L39

For reference, these are the flags use by Arch to compile C programs:

```bash
CFLAGS="-march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions \
        -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security \
        -fstack-clash-protection -fcf-protection \
        -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer"
```

Sapd, 2024-04-05:
Thanks for your input.

Also thanks for the compilation parameters @TheBill2001 
I will try to fix the resulting warnings for it. But I think I also have to get rid of the `-Werror` in the repo, as other Distributions can then again have other compilation parameters creating warnings and leading to an error.

For the asprintf Im not sure if I should replace it by the function in the code or check the return values. Checking the return values would be the 100% "right" way, but if something like memory allocation fails the program will crash anyway.

PR #337 (2024-04-06): Rename asprintf to _asprintf to avoid unused-result warnings
### Changes made

fixes #335

It would probably be best to actually check the return values for errors on writing. Maybe somethign along these lines:

```c
#define ASSERT_ASPRINTF(x) { if (x < 0) {fprintf(stderr, "Error formatting output"); }
```

### Checklist

- [x] I adjusted the README (if needed)
- [x] For new features in HeadsetControl: I discussed it beforehand in Issues or Discussions and adhered to the [wiki](https://github.com/Sapd/HeadsetControl/wiki/Development#adding-a-new-feature-to-the-headsestcontrol-application-itsel [... cut]
closed 2024-04-06 by Sapd