Sapd/HeadsetControl #336: Add support for Logitech G435

abdullah2993, 2024-04-03:
### Description

Add support for Logitech G435

### Headset Name

Logitech G435

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
Device Found
  VendorID: 0x046d
 ProductID: 0x0acb
  path: /dev/hidraw8
  serial_number: 202105190004
  Manufacturer: Logitech G series
  Product:      G435 Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x046d
 ProductID: 0x0acb
  path: /dev/hidraw8
  serial_number: 202105190004
  Manufacturer: Logitech G series
  Product:      G435 Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xb Usageid: 0x1

Device Found
  VendorID: 0x046d
 ProductID: 0x0acb
  path: /dev/hidraw8
  serial_number: 202105190004
  Manufacturer: Logitech G series
  Product:      G435 Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc0 Usageid: 0x1

Device Found
  VendorID: 0x046d
 ProductID: 0x0acb
  path: /dev/hidraw8
  serial_number: 202105190004
  Manufacturer: Logitech G series
  Product:      G435 Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc3 Usageid: 0x1

 ```

</details>

### New Device Implementation Checklist

- [X] I acknowledge that implementing a new device requires direct access to the device and familiarity with capturing and analyzing its communications. I have access to the device and am willing to contribute by capturing necessary data and/or attempting implementation.
- [ ] I have already attempted to capture device communications using WireShark or similar tools and am including my findings (successful or not) with this issue to assist in implementation efforts.
- [X] I am seeking guidance on how to begin this process and am willing to collaborate with the community to add support for this device. I understand where to find resources (refer to the WIKI).

levitskyyy, 2024-04-19:
this is very necessary

abdullah2993, 2024-04-24:
this one is kind of difficult since there is not driver for windows so its pretty hard to reverse engineer

Sapd, 2024-04-24:
> this one is kind of difficult since there is not driver for windows so its pretty hard to reverse engineer

How are you supposed to set then features like sidetone with this headset? Doesnt it come with a software?

abdullah2993, 2024-04-24:
no it doesn't. the side tone features are set directly via buttons on the headset which is quite annoying to remember.

abdullah2993, 2024-04-24:
> > this one is kind of difficult since there is not driver for windows so its pretty hard to reverse engineer
> 
> How are you supposed to set then features like sidetone with this headset? Doesnt it come with a software?

no it doesn't. the side tone features are set directly via buttons on the headset which is quite annoying to remember.

levitskyyy, 2024-04-24:
Is it possible to use this to show the charge of the device?

abdullah2993, 2024-04-24:
> Is it possible to use this to show the charge of the device?

I think it is possible since when connected to an android phone it shows the battery percentage.

abdullah2993, 2024-04-25:
> > > Is it possible to use this to show the charge of the device?
> > 
> > 
> > I think it is possible since when connected to an android phone it shows the battery percentage.
> 
> One good way to query this would be to sniff it somehow (WireShark has this capabilities also with Bluetooth, but may require a special bluetooth card or stick which can act as proxy).
> 
> Otherwise one could simply try to see if the device sends something by itself:
> 
> ```
> headsetcontrol --dev -- --device VENDORID:PRODUCTID --receive
> ```

the `--receive` method doesn't seem to work. From what I understand is that the battery level is only exposed via bluetooth and not via lightspeed, with lightspeed the headset only sends volume up and volume down and nothing else.

abdullah2993, 2025-03-04:
Closing this as it looks like the headphone does not support these controls over lightspeed

closed 2025-03-04 by abdullah2993