Sapd/HeadsetControl #339: Not showing correct battery status

devnore, 2024-04-04:
### Description

Hi Denis!

First of all, it is awesome to see the formatting (yaml/json/env) added 

But I ran into an issue with the output that I guess is specific to either my specific model or Steelseries in general as the functionality  was discussed in #271 and seemed to be working with Corsair Void.

The issue:
My battery status is always showing `BATTERY_AVAILABLE` regardless of if it is charging/discharging/off the only value that changes (when turned off is the battery level which reports 0 (and that should be fine)

On, Connected/Charging
``` 
name: "HeadsetControl"
version: "3.0.0-1-g0005a0a"
api_version: "1.0"
hidapi_version: "0.11.2"
device_count: 1
devices:
  - status: "success"
    device: "SteelSeries Arctis (7/Pro)"
    vendor: "SteelSeries "
    product: "SteelSeries Arctis 7"
    id_vendor: "0x1038"
    id_product: "0x12ad"
    capabilities:
      - CAP_SIDETONE
      - CAP_BATTERY_STATUS
      - CAP_LIGHTS
      - CAP_INACTIVE_TIME
      - CAP_CHATMIX_STATUS
    capabilities_str:
      - sidetone
      - battery
      - lights
      - inactive time
      - chatmix
    battery:
      status: "BATTERY_AVAILABLE"
      level: 70
    chatmix: 53
```

Off, Connected/Charging
```name: "HeadsetControl"
version: "3.0.0-1-g0005a0a"
api_version: "1.0"
hidapi_version: "0.11.2"
device_count: 1
devices:
  - status: "success"
    device: "SteelSeries Arctis (7/Pro)"
    vendor: "SteelSeries "
    product: "SteelSeries Arctis 7"
    id_vendor: "0x1038"
    id_product: "0x12ad"
    capabilities:
      - CAP_SIDETONE
      - CAP_BATTERY_STATUS
      - CAP_LIGHTS
      - CAP_INACTIVE_TIME
      - CAP_CHATMIX_STATUS
    capabilities_str:
      - sidetone
      - battery
      - lights
      - inactive time
      - chatmix
    battery:
      status: "BATTERY_AVAILABLE"
      level: 0
    chatmix: 64
```

Off, Disconnected
``` 
---
name: "HeadsetControl"
version: "3.0.0-1-g0005a0a"
api_version: "1.0"
hidapi_version: "0.11.2"
device_count: 1
devices:
  - status: "success"
    device: "SteelSeries Arctis (7/Pro)"
    vendor: "SteelSeries "
    product: "SteelSeries Arctis 7"
    id_vendor: "0x1038"
    id_product: "0x12ad"
    capabilities:
      - CAP_SIDETONE
      - CAP_BATTERY_STATUS
      - CAP_LIGHTS
      - CAP_INACTIVE_TIME
      - CAP_CHATMIX_STATUS
    capabilities_str:
      - sidetone
      - battery
      - lights
      - inactive time
      - chatmix
    battery:
      status: "BATTERY_AVAILABLE"
      level: 0
    chatmix: 64
```

### Headset Name

SteelSeries Arctis 7

### On which OS does the problem happen?

Linux

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
$>headsetcontrol --dev -- --list --device 0x1038:0x12ad
Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff43 Usageid: 0x202

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xc

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0x1 Usageid: 0x3a

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xc

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw10
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0x1 Usageid: 0x3a
 ```

</details>

Sapd, 2024-04-05:
The Steelseries implementation does not define that, thats why it always show BATTERY_AVAILABLE, see here: https://github.com/Sapd/HeadsetControl/blob/0005a0a5fd9c27b02d8fb49f138a34cd2ddd4c78/src/devices/steelseries_arctis_7.c#L82-L123

Basically, BATTERY_AVAILABLE must be seen as general status, it just means that the Battery info could be read from the device. (and in this case it reads as 0). Probably there is another flag to check if it is charging, but you would have to take a look with WireShark. Or alternatively and easier, you can try this:
```
headsetcontrol --dev -- --device 0x1038:0x12ad --send 0x06,0x18 --receive
```
then see if one of the bytes change during charging. You could then simply supplement the value in the code and open up a PR.

devnore, 2024-04-09:
I currently don't have my GitHub creds set up on my work-laptop (so unable to push). But here is my quick and dirty solution 

```
diff --git a/src/devices/steelseries_arctis_7.c b/src/devices/steelseries_arctis_7.c
index bd52730..a86ce04 100644
--- a/src/devices/steelseries_arctis_7.c
+++ b/src/devices/steelseries_arctis_7.c
@@ -112,13 +112,16 @@ static BatteryInfo arctis_7_request_battery(hid_device* device_handle)
 
     int bat = data_read[2];
 
-    info.status = BATTERY_AVAILABLE;
+    // 0% battery == BATTERY_UNAVAILABLE
+    if (bat == 0)
+        return info;
 
     if (bat > 100)
         info.level = 100;
     else
         info.level = bat;
 
+    info.status = BATTERY_AVAILABLE;
     return info;
 }
```

closed 2025-04-05 by github-actions[bot]