Sapd/HeadsetControl #340: Can not execute any command without sudo privileges

CommanderData1701, 2024-04-06:
### Description

Hello, I tried fetching my Headset's battery status (Corsair HS 70 pro) in order to display that information in my instance of xmobar. I am running arch with kernel version 6.8.2. 
If I try running `headsetcontrol -b` or with any other option (without sudo) I get the output:
```
Found Corsair Headset Device!

Error: [battery] Could not open device. Error: Failed to open a device with path '/dev/hidraw0': Permission denied
```
The vendor id and product id are listed int the `.rules` file at `/usr/local/lib/udev/rules.d/70-headsets.rules`, and I did try reloading these rules with the command given in the readme and  rebooting. What might be the issue here?

### Headset Name

Corsair HS 70 Pro

### On which OS does the problem happen?

Linux

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
 Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0x64

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0x65

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0x66

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0xc8

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0xc9

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0xca

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0xd9

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0xcb

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0xd3

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xffc5 Usageid: 0xcc

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xc

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0x1 Usageid: 0x3a

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0xff00 Usageid: 0xc

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a4f
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: Corsair
  Product:      CORSAIR HS70 Pro Wireless Gaming Headset
  Interface:    3
  Usage-Page: 0x1 Usageid: 0x3a

Device Found
  VendorID: 0x093a
 ProductID: 0x2521
  path: /dev/hidraw3
  serial_number: 
  Manufacturer: (null)
  Product:      USB OPTICAL MOUSE
  Interface:    0
  Usage-Page: 0x1 Usageid: 0x2

Device Found
  VendorID: 0x093a
 ProductID: 0x2521
  path: /dev/hidraw3
  serial_number: 
  Manufacturer: (null)
  Product:      USB OPTICAL MOUSE
  Interface:    0
  Usage-Page: 0x1 Usageid: 0x1

Device Found
  VendorID: 0x258a
 ProductID: 0x003a
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: SINO WEALTH
  Product:      Gaming KB 
  Interface:    0
  Usage-Page: 0x1 Usageid: 0x6

Device Found
  VendorID: 0x258a
 ProductID: 0x003a
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: SINO WEALTH
  Product:      Gaming KB 
  Interface:    1
  Usage-Page: 0x1 Usageid: 0x80

Device Found
  VendorID: 0x258a
 ProductID: 0x003a
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: SINO WEALTH
  Product:      Gaming KB 
  Interface:    1
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x258a
 ProductID: 0x003a
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: SINO WEALTH
  Product:      Gaming KB 
  Interface:    1
  Usage-Page: 0xff00 Usageid: 0x1

Device Found
  VendorID: 0x258a
 ProductID: 0x003a
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: SINO WEALTH
  Product:      Gaming KB 
  Interface:    1
  Usage-Page: 0x1 Usageid: 0x6

 ```

</details>

Sapd, 2024-04-12:
Maybe you could try to move them to one of the following folders: https://wiki.archlinux.org/title/udev#About_udev_rules

CommanderData1701, 2024-04-12:
I did try that, with no success unfortunately. I am really confused, because it used to work in the past... I think before I used the version from the [AUR](https://aur.archlinux.org/packages/headsetcontrol), and not building from source. The issue here is that things are really hard to reproduce, since Arch installs mostly are not the same

closed 2025-04-08 by github-actions[bot]