Sapd/HeadsetControl #346: Support for HyperX Cloud III (with wire)

AdamGardelov, 2024-05-08:
### Description

I would really appreciate if could add support for the wired version of HyperX Cloud III. 

### Headset Name

HyperX Cloud III

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
Device Found
  VendorID: 0x03f0
 ProductID: 0x089d
  path: /dev/hidraw6
  serial_number: 000000000000
  Manufacturer: HP, Inc
  Product:      HyperX Cloud III
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x03f0
 ProductID: 0x089d
  path: /dev/hidraw6
  serial_number: 000000000000
  Manufacturer: HP, Inc
  Product:      HyperX Cloud III
  Interface:    3
  Usage-Page: 0x1c0 Usageid: 0x1

Device Found
  VendorID: 0x03f0
 ProductID: 0x089d
  path: /dev/hidraw6
  serial_number: 000000000000
  Manufacturer: HP, Inc
  Product:      HyperX Cloud III
  Interface:    3
  Usage-Page: 0x170f Usageid: 0x1

Device Found
  VendorID: 0x03f0
 ProductID: 0x089d
  path: /dev/hidraw6
  serial_number: 000000000000
  Manufacturer: HP, Inc
  Product:      HyperX Cloud III
  Interface:    3
  Usage-Page: 0xffc0 Usageid: 0x1

 ```

</details>

### New Device Implementation Checklist

- [X] I acknowledge that implementing a new device requires direct access to the device and familiarity with capturing and analyzing its communications. I have access to the device and am willing to contribute by capturing necessary data and/or attempting implementation.
- [X] I have already attempted to capture device communications using WireShark or similar tools and am including my findings (successful or not) with this issue to assist in implementation efforts.
- [X] I am seeking guidance on how to begin this process and am willing to collaborate with the community to add support for this device. I understand where to find resources (refer to the WIKI).

AdamGardelov, 2024-05-08:
This is my finding from Wireshark. 
`ON

00000000  20 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . .  
0000002E  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00                                                                                            . . . . . . . . . . . . . . . .                                                              

OFF

00000000  20 86 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00     . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . .  
0000002E  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

Sapd, 2024-05-08:
The headers are missing. What does WireShark tell you on the "Setup Data" section?

AdamGardelov, 2024-05-08:
This is off:

[image]

This is on, witch makes 2 requests:

1.
[image]
2. 
[image]

Sapd, 2024-05-08:
I think this should work, you can try:

```
headsetcontrol --dev -- --device 0x03f0:0x089d  --send-feature "0x20 0x05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"
```

and

```
headsetcontrol --dev -- --device 0x03f0:0x089d  --send-feature "0x20 0x86 0x01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"
```

AdamGardelov, 2024-05-08:
Sadly this gives "Couldn't open device."

AdamGardelov, 2024-05-08:
Thanks, running as root worked. But none of the commands turned sideton off unfortunately

Sapd, 2024-05-08:
I didnt see that you had 3 payloads there.

What about

```
headsetcontrol --dev -- --device 0x03f0:0x089d  --send-feature "0x20 0x86 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"
```

AdamGardelov, 2024-05-09:
It worked!! if you want to add this device to the repo:

This is On:
```
headsetcontrol --dev -- --device 0x03f0:0x089d  --send-feature "0x20 0x86 0x01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"
```

And this is off:

```
headsetcontrol --dev -- --device 0x03f0:0x089d  --send-feature "0x20 0x86 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"
```

Thanks for your help man, really appreciate it! Support 10/10!

Sapd, 2024-05-10:
I included it in the hyperx branch: https://github.com/Sapd/HeadsetControl/pull/348. Can you try it?

First delete the build folder: `rm -r build`
Then:
```
git pull
git checkout hyperx
mkdir build
cd build
cmake ..
make
```

AdamGardelov, 2024-05-13:
Works perfect! Thanks!

closed 2024-05-13 by AdamGardelov