Sapd/HeadsetControl #347: Arctis 7 Pro - Battery Charge Level alway at 56%

dmatta22201, 2024-05-09:
### Description

I am having an issue in which the Arctis 7 Pro is always reported with a battery charge of 56%, regardless of actual battery level. Running

```headsetcontrol --dev -- --device 0x1038:0x12ad --send 0x06,0x18 --receive```

The response is consistently:

```0x06 0x18 0x38 ...```

where the ... is 6 bytes whose values vary, followed by a bunch of zero value bytes.

Looking at /devices/steelseries_arctis_7.c in the arctis_7_request_battery function, it looks like it is setting the battery level to the third byte...

```
    int bat = data_read[2];

    info.status = BATTERY_AVAILABLE;

    if (bat > 100)
        info.level = 100;
    else
        info.level = bat;
```

However the third byte is always 0x38 (decimal 56).

I am not sure how to interpret the next 6 bytes returned. Rerunning the command every 30 seconds or so produced the following:

```
0000 0000 0000 0000 0000 0000
0000 0xff 0000 0x06 0xc0 0000
0000 0xff 0000  0x06 0xc0 0000
0000 0xff 0000  0x06 0xc0 0xc0
```
Does anyone have a reference of what these bytes are and if they relate to battery levels?

### Headset Name

Steelseries Arctis 7/Pro

### On which OS does the problem happen?

Linux

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
 Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff43 Usageid: 0x202

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x22

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xa

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xc

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0x1 Usageid: 0x3a

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0xc

Device Found
  VendorID: 0x1038
 ProductID: 0x12ad
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries 
  Product:      SteelSeries Arctis 7
  Interface:    5
  Usage-Page: 0x1 Usageid: 0x3a

 ```

</details>

ChrisLauinger77, 2024-11-02:
I updated from a Arctis 7 (USC-A) to Arctis 7+ (USB-C)
Since then the battery level is reported as 50 for me. The indicator light on the headset is yellow (below 50%)
I use self-compiled latest version from master.
```
{
  "name": "HeadsetControl",
  "version": "3.0.0-23-gc0c9b50",
  "api_version": "1.2",
  "hidapi_version": "0.14.0",
  "device_count": 1,
  "devices": [
    {
      "status": "success",
      "device": "SteelSeries Arctis 7+",
      "vendor": "SteelSeries",
      "product": "Arctis 7+",
      "id_vendor": "0x1038",
      "id_product": "0x220e",
      "capabilities": [
        "CAP_SIDETONE", 
        "CAP_BATTERY_STATUS", 
        "CAP_INACTIVE_TIME", 
        "CAP_CHATMIX_STATUS", 
        "CAP_EQUALIZER_PRESET", 
        "CAP_EQUALIZER"
      ],
      "capabilities_str": [
        "sidetone", 
        "battery", 
        "inactive time", 
        "chatmix", 
        "equalizer preset", 
        "equalizer"
      ],
      "battery": {
        "status": "BATTERY_AVAILABLE",
        "level": 50
      },
      "equalizer": {
        "bands": 10,
        "baseline": 0,
        "step": 0.5,
        "min": -12,
        "max": 12
      },
      "chatmix": 64
    }
  ]
}
```

ChrisLauinger77, 2024-11-02:
I charged it to 100% - now it shows also 100
Lets see If it stuck at 50 again as soon as its discharged ...

```
{
  "name": "HeadsetControl",
  "version": "3.0.0-23-gc0c9b50",
  "api_version": "1.2",
  "hidapi_version": "0.14.0",
  "device_count": 1,
  "devices": [
    {
      "status": "success",
      "device": "SteelSeries Arctis 7+",
      "vendor": "SteelSeries",
      "product": "Arctis 7+",
      "id_vendor": "0x1038",
      "id_product": "0x220e",
      "capabilities": [
        "CAP_SIDETONE", 
        "CAP_BATTERY_STATUS", 
        "CAP_INACTIVE_TIME", 
        "CAP_CHATMIX_STATUS", 
        "CAP_EQUALIZER_PRESET", 
        "CAP_EQUALIZER"
      ],
      "capabilities_str": [
        "sidetone", 
        "battery", 
        "inactive time", 
        "chatmix", 
        "equalizer preset", 
        "equalizer"
      ],
      "battery": {
        "status": "BATTERY_AVAILABLE",
        "level": 100
      },
      "equalizer": {
        "bands": 10,
        "baseline": 0,
        "step": 0.5,
        "min": -12,
        "max": 12
      },
      "chatmix": 64
    }
  ]
}
```

polgarc, 2025-07-13:
I'm not 100% sure but I think the issue is that the device only updates the battery % when it's charging (plugged in) **and** it's also turned on (and maybe has to be connected to the hub too?). For example in my case:

1. headset's battery died (0%)
2. I charged it to 100% with a wall plug while it was off and also wasn't connected to the hub either (PC was off)
3. turned on my PC, headset works just fine
4. ran the tool, it always reported 0% - and even my led started blinking red like it was empty
5. plugged in to the PC with a USB cable
6. battery % went 0% -> 38% -> 80% -> 96% within ~10 seconds as I was running `headsetcontrol -b` again and again

So I think this is simply poor reporting on the headset's side. I'll check how the number's going to change when I use it more.

BTW mine is:

```yaml
    device: "SteelSeries Arctis (7/Pro)"
    vendor: "SteelSeries "
    product: "SteelSeries Arctis 7"
    id_vendor: "0x1038"
    id_product: "0x12ad"
```