Sapd/HeadsetControl #350: version 3.0.0 build failure in src/output.c with CMake

mattmcadoo, 2024-05-19:
### Description

`
In function ‘handle_feature’,
    inlined from ‘main’ at /var/tmp/portage/app-misc/headsetcontrol-3.0.0/work/headsetcontrol-3.0.0/src/main.c:860:45:
/var/tmp/portage/app-misc/headsetcontrol-3.0.0/work/headsetcontrol-3.0.0/src/main.c:314:8: error: ‘ret’ may be used uninitialized [-Werror=maybe-uninitialized]
  314 |     if (ret >= 0) {
      |        ^
/var/tmp/portage/app-misc/headsetcontrol-3.0.0/work/headsetcontrol-3.0.0/src/main.c: In function ‘main’:
/var/tmp/portage/app-misc/headsetcontrol-3.0.0/work/headsetcontrol-3.0.0/src/main.c:215:9: note: ‘ret’ was declared here
  215 |     int ret;
      |         ^~~
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/headsetcontrol.dir/build.make:76: CMakeFiles/headsetcontrol.dir/src/main.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/headsetcontrol.dir/build.make:146: CMakeFiles/headsetcontrol.dir/src/output.c.o] Error 1
make[2]: Leaving directory '/var/tmp/portage/app-misc/headsetcontrol-3.0.0/work/headsetcontrol-3.0.0_build'
make[1]: *** [CMakeFiles/Makefile2:149: CMakeFiles/headsetcontrol.dir/all] Error 2
make[1]: Leaving directory '/var/tmp/portage/app-misc/headsetcontrol-3.0.0/work/headsetcontrol-3.0.0_build'
make: *** [Makefile:146: all] Error 2
`

I'm using a very basic ebuild file that was based up an older version of headsetcontrol in a 3rd party overlay for Gentoo.

Portage 3.0.64 (python 3.11.9-final-0, default/linux/amd64/23.0/desktop/gnome/systemd, gcc-13, glibc-2.39-r6, 6.9.0-gentoo-x86_64 x86_64)

sh bash 5.2_p26-r3
ld GNU ld (Gentoo 2.42 p3) 2.42.0
app-misc/pax-utils:        1.3.7::gentoo
app-shells/bash:           5.2_p26-r3::gentoo
dev-build/autoconf:        2.13-r8::gentoo, 2.71-r6::gentoo, 2.72-r1::gentoo
dev-build/automake:        1.16.5-r2::gentoo
dev-build/cmake:           3.29.3::gentoo
dev-build/libtool:         2.4.7-r4::gentoo
dev-build/make:            4.4.1-r1::gentoo
dev-build/meson:           1.4.0-r1::gentoo
dev-lang/python:           3.10.14_p1::gentoo, 3.11.9::gentoo, 3.12.3::gentoo, 3.13.0_beta1_p1::gentoo
dev-lang/rust:             1.77.1::gentoo
sys-apps/baselayout:       2.15::gentoo
sys-apps/sandbox:          2.38::gentoo
sys-apps/systemd:          255.6::gentoo
sys-devel/binutils:        2.41-r5::gentoo, 2.42-r1::gentoo
sys-devel/binutils-config: 5.5::gentoo
sys-devel/clang:           16.0.6::rocm-bleeding-edge, 17.0.6::rocm-bleeding-edge, 18.1.5::gentoo
sys-devel/gcc:             13.2.1_p20240503::gentoo
sys-devel/gcc-config:      2.11::gentoo
sys-devel/lld:             17.0.6::gentoo, 18.1.5::gentoo
sys-devel/llvm:            16.0.6::gentoo, 17.0.6::rocm-bleeding-edge, 18.1.5-r1::gentoo
sys-kernel/linux-headers:  6.9::gentoo (virtual/os-headers)
sys-libs/glibc:            2.39-r6::gentoo
dev-libs/hidapi:        0.14.0::gentoo

CBUILD="x86_64-pc-linux-gnu"
CFLAGS="-march=native -O2 -pipe"
CHOST="x86_64-pc-linux-gnu"
CXXFLAGS="-march=native -O2 -pipe"
FCFLAGS="-O2 -pipe"
FFLAGS="-O2 -pipe"
LDFLAGS="-Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs"

### On which OS does the problem happen?

Linux

Sapd, 2024-05-29:
Yeah its a similar problem to: https://github.com/Sapd/HeadsetControl/issues/335

Did you use the latest version from Git? Actually I removed there that it should treat warnings as error. But could be that your OS somehow still injects -Werror

mattmcadoo, 2024-06-01:
No, I was only using the released version.  Are you planning on doing a 3.0.1 soon or planning for a bigger one further away?

Sapd, 2024-06-09:
> No, I was only using the released version. Are you planning on doing a 3.0.1 soon or planning for a bigger one further away?

I will probably make one soon, but its always better with this project to use the master (as stated in the readme).

mattmcadoo, 2025-04-30:
Any update on cutting a new release version?

nicola02nb, 2025-05-06:
@mattmcadoo If you need latest build you can download the `continuous build` `pre-release` at [Sapd/HeadsetControl/releases](https://github.com/Sapd/HeadsetControl/releases)

closed 2025-05-22 by Sapd