Sapd/HeadsetControl #351: CORSAIR HS80 RGB Wireless Gaming Support

VictorAseo, 2024-05-24:
### Description

Could you please add this one? I'm experiencing constant issues with mic loopback every time I turn on my microphone. I'm unsure which device ID is the real one.

[image]

### Headset Name

CORSAIR HS80 RGB Wireless Gaming

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
 Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a6b
  path: /dev/hidraw5
  serial_number: 16d64dc300050215
  Manufacturer: Corsair
  Product:      CORSAIR HS80 RGB Wireless Gaming Receiver
  Interface:    3
  Usage-Page: 0xff40 Usageid: 0x6

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a6b
  path: /dev/hidraw5
  serial_number: 16d64dc300050215
  Manufacturer: Corsair
  Product:      CORSAIR HS80 RGB Wireless Gaming Receiver
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a6b
  path: /dev/hidraw5
  serial_number: 16d64dc300050215
  Manufacturer: Corsair
  Product:      CORSAIR HS80 RGB Wireless Gaming Receiver
  Interface:    3
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a6b
  path: /dev/hidraw5
  serial_number: 16d64dc300050215
  Manufacturer: Corsair
  Product:      CORSAIR HS80 RGB Wireless Gaming Receiver
  Interface:    3
  Usage-Page: 0xff42 Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a6b
  path: /dev/hidraw5
  serial_number: 16d64dc300050215
  Manufacturer: Corsair
  Product:      CORSAIR HS80 RGB Wireless Gaming Receiver
  Interface:    3
  Usage-Page: 0xff42 Usageid: 0x2

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a6b
  path: /dev/hidraw5
  serial_number: 16d64dc300050215
  Manufacturer: Corsair
  Product:      CORSAIR HS80 RGB Wireless Gaming Receiver
  Interface:    3
  Usage-Page: 0xff58 Usageid: 0x1

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a6b
  path: /dev/hidraw6
  serial_number: 16d64dc300050215
  Manufacturer: Corsair
  Product:      CORSAIR HS80 RGB Wireless Gaming Receiver
  Interface:    4
  Usage-Page: 0xff41 Usageid: 0x2

Device Found
  VendorID: 0x1b1c
 ProductID: 0x0a6b
  path: /dev/hidraw6
  serial_number: 16d64dc300050215
  Manufacturer: Corsair
  Product:      CORSAIR HS80 RGB Wireless Gaming Receiver
  Interface:    4
  Usage-Page: 0xff41 Usageid: 0x1
 ```

</details>

### New Device Implementation Checklist

- [X] I acknowledge that implementing a new device requires direct access to the device and familiarity with capturing and analyzing its communications. I have access to the device and am willing to contribute by capturing necessary data and/or attempting implementation.
- [ ] I have already attempted to capture device communications using WireShark or similar tools and am including my findings (successful or not) with this issue to assist in implementation efforts.
- [ ] I am seeking guidance on how to begin this process and am willing to collaborate with the community to add support for this device. I understand where to find resources (refer to the WIKI).

Sapd, 2024-05-24:
https://github.com/Sapd/HeadsetControl/issues/178

you should be able to control it with AlsaMixer or similar software

chisenberg, 2024-11-16:
Any chance the RGB control is done via HID and could be implemented on HeadsetControle ? I know the sidetone must me changed using Alsamixer, but the battery drains too fast with RGB on.

dreua, 2025-02-15:
> but the battery drains too fast with RGB on

I turned that off once in Windows and it stayed off so far. Would be cool to have Linux support as well, of course.