Sapd/HeadsetControl #353: Add Volume Limiter Support

duckman, 2024-06-03:
### Description

Would love to be able to disable the volume limiter on the SteelSeries Arctis Nova 7. Looks like there is a function in the device file but I don't see a way to execute it.

### Headset Name

SteelSeries Arctis Nova 7

Sapd, 2024-06-09:
> Looks like there is a function in the device file but I don't see a way to execute it.

Yeah the guy implementing it found it probably during his WireShark Session but didn't implement it as headsetcontrol currently has no command for that.

You can feel free to add one (not really complicated, you can just copy the code for send_microphone_volume):

- Add here a new funtion, similar to `send_microphone_volume`: https://github.com/Sapd/HeadsetControl/blob/f64ac48384cbb7df3878749cbe9635adefdae84c/src/device.h#L324
- In the Arctis 7 file, reference the function you saw here: https://github.com/Sapd/HeadsetControl/blob/f64ac48384cbb7df3878749cbe9635adefdae84c/src/devices/steelseries_arctis_7.c#L40
- Also in device.h add a capbability just calling it `CAP_VOLUME_LIMITER` : https://github.com/Sapd/HeadsetControl/blob/f64ac48384cbb7df3878749cbe9635adefdae84c/src/device.h#L38
- Now you just need to implement the parameter in main.c:
     - Add an option here `{ "volume-limiter", required_argument, NULL, 0 }, `https://github.com/Sapd/HeadsetControl/blob/f64ac48384cbb7df3878749cbe9635adefdae84c/src/main.c#L529
     - Handle that option here, simply copy the microphone volume code and rename things (and create a variable on top): https://github.com/Sapd/HeadsetControl/blob/f64ac48384cbb7df3878749cbe9635adefdae84c/src/main.c#L706-L712
     - Also here: https://github.com/Sapd/HeadsetControl/blob/f64ac48384cbb7df3878749cbe9635adefdae84c/src/main.c#L806
     - Then call your new function in handle feature: https://github.com/Sapd/HeadsetControl/blob/f64ac48384cbb7df3878749cbe9635adefdae84c/src/main.c#L301
    

Feel free if you struggle somewhere

nicola02nb, 2024-06-10:
I created a [pull request](https://github.com/Sapd/HeadsetControl/pull/356) for some missing features of the `Steelseries Artic Nova 7`:
- microphone mute led brightness
- microphone volume
- volume limiter

duckman, 2024-06-10:
You sir are a gentleman and a scholar. I'll try it out tonight.

nicola02nb, 2024-06-10:
If you want to checck out also [my GUI](https://github.com/nicola02nb/HeadsetControl-GUI/tree/main) you're welcome.

closed 2024-06-11 by Sapd