Sapd/HeadsetControl #354: Add Eqalizer info to json output

nicola02nb, 2024-06-06:
### Description

Add for devices that supports the equalizer `-e --equalizer` argument some info when calling `-o json` like `EQUALIZER_BANDS_SIZE`, `EQUALIZER_BAND_MIN`, `EQUALIZER_BAND_MAX`.
It can be a sectionlike this:
``` json
{
  "name": "HeadsetControl",
  "devices": [
    {
      "status": "success",
      "device": "HeadsetControl Test device",
      "vendor": "HeadsetControl",
      "product": "Test device",
      "id_vendor": "0xf00b",
      "id_product": "0xa00c",
      "capabilities": [
        "CAP_EQUALIZER",
      ],
      "capabilities_str": [
        "equalizer"
      ],
      "equalizer": {
        "band_size": 10
        "min": -10,
        "max": 10
      },
    }
  ]
}
```

nicola02nb, 2024-06-08:
It would be also cool a section `equalizer_preset` that returns the presets of the headpones:
``` json
euqalizer_preset[
   flat: [0,....0,],
   bass: [....],
   ....
]
```

Sapd, 2024-06-09:
Yeah that is quite easy to do.

However I do not fully understand your preset example. So basically what I see what we can return is something like this:
```
euqalizer_preset[
   flat,
   bass,
   smiley,
   focus
]
```

Where are the numbers in your example come from? (can you also link me the device file when you reference a specific one)

nicola02nb, 2024-06-09:
Yeah, sorrry I missed some information...
I was thinking something like this:
``` json
"euqalizer_preset":[
   "flat":[0,0,0,0,0,0,0],
   "bass":[0,0,0,0,0,0,0],
   "smiley":[0,0,0,0,0,0,0],
   "focus":[0,0,0,0,0,0,0],
]
```
I'm writing a GUI for HeadsetControl, and having those informations would be useful.

Sapd, 2024-06-09:
Ah yes now I get it, I saw the code before didn't realize that the command actually simply contains equalizer-payload. 

Probably needs some code cleanup with more data-driven approach (defining the equalizer as struct and also using it to directly pass it to hid_write), but otherwise easily doable. 

I think it can be implemented as a device function, which simply returns that information somewhere here - similar to other request functions (but that it instead returns the data from variables instead of from HID):
https://github.com/Sapd/HeadsetControl/blob/f64ac48384cbb7df3878749cbe9635adefdae84c/src/device.h#L235

Can do so in the near future

nicola02nb, 2024-06-09:
Yeah, it could be an idea... it seems also quite simply to implement...
Tell me if you need anithing else

nicola02nb, 2024-06-16:
@Sapd  I done a [commit](https://github.com/Sapd/HeadsetControl/commit/de055420c3d12d4ef106888712c962ed748c01df) on my fork where I added the feature for json output... It needs some refactor of some comments on `device.h` added function firms, and it needs to implement all missing functions in the compatible devices... can you give a look at it?

Sapd, 2024-06-17:
Some suggestions:
- Maybe it would be easier to be a device information in `device.h` in the device struct instead of a "feature"
    - As when you request json output without any options you will get all information anyway. And this information is static in any case
    - `EqualizerInfo` would have to be a pointer then inside the device struct, which is simply `NULL` when there is no EQ info. When it is null no output is needed
    - You would also then do not need the functions like `request_equalizer`

nicola02nb, 2024-06-17:
Did a rollback and a [new commit](https://github.com/Sapd/HeadsetControl/compare/master...nicola02nb:HeadsetControl:master) following your idea.

Sapd, 2024-06-19:
Looks fine to me on first looks, in any case you can create a PR from your fork

closed 2024-08-01 by Sapd