Sapd/HeadsetControl #359: SteelSeries Arctis Nova 5

fatihmete, 2024-06-27:
### Description

Add support for SteelSeries Arctis Nova 5

### Headset Name

SteelSeries Arctis Nova 5

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
Device Found
  VendorID: 0x1038
 ProductID: 0x2232
  path: /dev/hidraw7
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis Nova 5
  Interface:    3
  Usage-Page: 0xffc0 Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x2232
  path: /dev/hidraw8
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis Nova 5
  Interface:    4
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x2232
  path: /dev/hidraw9
  serial_number: 
  Manufacturer: SteelSeries
  Product:      SteelSeries Arctis Nova 5
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x1

 ```

</details>

### New Device Implementation Checklist

- [X] I acknowledge that implementing a new device requires direct access to the device and familiarity with capturing and analyzing its communications. I have access to the device and am willing to contribute by capturing necessary data and/or attempting implementation.
- [ ] I have already attempted to capture device communications using WireShark or similar tools and am including my findings (successful or not) with this issue to assist in implementation efforts.
- [X] I am seeking guidance on how to begin this process and am willing to collaborate with the community to add support for this device. I understand where to find resources (refer to the WIKI).

ninelore, 2024-07-02:
hi @fatihmete, the nova 5 specifically has a phone companion app with which you can set all settings of the headset, which you can use instead or in the meantime

fatihmete, 2024-07-03:
@ninelore thank you for the suggestion.  I'm particularly interested in monitoring battery status and changing equalizer presets from the PC. Currently, I'm following the developer wiki page. I discovered the sidetone command using Wireshark documentation.

Code below set sidetone to 0:
`sudo ./headsetcontrol --dev -- --device 0x1038:0x2232 --send-feature "
0x39, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
0x00, 0x00, 0x00, 0x00"`

To set sidetone to 5, modify the 2nd value as follows:
`0x39, 0x5, 0x00, ....`

This is my first experience working with USB communication. I hope to create a pull request for Nova 5 soon.

hrzlgnm, 2024-07-19:
Sorry, i didn't read through the issue and didn't see the device request. I' got stuff working with #361

Sapd, 2024-07-22:
Fixed in #361

Feel free to reopen if something is missing

closed 2024-07-22 by Sapd