Sapd/HeadsetControl #363: Logitech G935 Equalizer Support

Tom-jm03, 2024-08-02:
### Description

Hey,

i would really like to use the equalizer on linux, because the headset sounds like trash otherwise.

May i ask, why it hasn't been implemented yet? Is it because it hasn't been reverse engineered?

If so, i can provide specific device details etc. 

I found out that the equalizer gets parsed via json under and lays under `Program Files/LGHUB/data/eq_presets.json` and `pro_eq_presets.json`.

It looks something like this:

```json   {
      "id": "b4bdd01f-a3a8-4163-817e-43808ffbf4bf",
      "name": "G2 shox",
      "attribute": "PRO_EQUALIZER",
      "applicationId": "",
      "profileId": "",
      "readOnly": true,
      "category": "",
      "proEqualizer": {
        "bands": [
          1,
          -1,
          -2,
          1,
          2
        ],
        "deviceType": "HEADSET",
        "type": "PRO",
        "name": "G2 shox",
        "tag": "PRO_PRESET"
      }

### Headset Name

Logitech G935

Sapd, 2024-08-03:
Hello,

> May i ask, why it hasn't been implemented yet? Is it because it hasn't been reverse engineered?

Yes. You would have to use WireShark to find out how the packets work: https://github.com/Sapd/HeadsetControl/wiki/Development-1-%E2%80%90-Adding-a-device#adding-a-completely-new-headset

Tom-jm03, 2024-08-03:
Thank you :)

I will try my Luck.

Tom-jm03, 2024-08-03:
I'm some steps further now.

If you select a Equalizer in the GHUB Software, nothing is being sent between the Headset/Host so it seems like that the Equalizer is Software Side?

But I found out something else.

When you select `Headset` -> `Settings in the bottom left corner` -> `On-Board Equalizer` you have the option to save a Equalizer on the Device itself and that actually did send something over USB:

[image]

So i will try now to manually send this data back. 
But there is something i don't quite understand.
The max Volume on Linux is way lower compared to Windows. But i don't know why exactly..

Maybe you have a clue?

Tom-jm03, 2024-08-03:
Hey,

i got a error when trying to send some Data.

Am i doing something wrong?

`./headsetcontrol --dev -- --device 0x046d:0x0a87 --send-feature --receive "
0x1c, 0x00, 0x90, 0x92, 0xfd, 0x9c, 0x8c, 0xe4, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x10, 0x00,
0x1a, 0x00, 0x00, 0x00, 0x00, 0x02, 0x1c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x03, 0x00, 0x00, 0x00, 0x11, 0xff, 0x07, 0x1a, 0x1f, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00"
`

The error i get after running:

`Failed to send data. Error: -1: ioctl (SFEATURE): Broken pipe`
I tried to get the battery level from the headset and it worked (with the -b option).

I haven't really worked with USB before, so pardon me, if I'm over seeing something.

Sapd, 2024-08-03:
The order of your parameters is mismatched. For some reason you put in the data after `--receive` which cannot work, as the data has to belong to the send command.

Also instead of send-feature you probably should try `--send THEDATA`

closed 2025-01-31 by tom-jm69