Sapd/HeadsetControl #365: GUI for HeadsetControl

nicola02nb, 2024-08-20:
Hi @Sapd,
in last months I developed a GUI for HeadsetControl, do you want to [give it a look](https://github.com/nicola02nb/HeadsetControl-GUI) and maybe add to your readme as a third party program/script?

Sapd, 2024-08-20:
Hello,

feel free to create a PR for including it

nicola02nb, 2024-08-20:
It's actually a separate program that uses the exe from your gh release.
It's a total different project take a look there: [https://github.com/nicola02nb/HeadsetControl-GUI](https://github.com/nicola02nb/HeadsetControl-GUI)

Sapd, 2024-08-20:
Yep, I already took a deep look, nice work :)

My answer refers to the readme, you can simply create a PR here to link your project in the readme

nicola02nb, 2024-08-20:
Oh, ok...  sorry for misunderstanding

closed 2024-08-26 by nicola02nb