Sapd/HeadsetControl #372: Incorrect battery levels for SteelSeries Arctis Nova 7 Wireless

ProLejka, 2024-09-27:
### Description

Hello. I have weird bug. When I check battery levels on my Arctis Nova 7 there seems to be only 3 states 100% 50% and 5%. I cross checked with app in Windows and it looks like 100% is 100%. But then it gets weird.
In Linux headsetcontrol reports 50% when there is battery in official app between 100% and 50%.
And after that it gets weirder because if it's under 50% (in official app) it just reports 5% in Linux. You can even see it on led on headphones that it reports for example 50% - orange led. If it's really 5% it would be red. It's very confusing because now I still have to check led on headphones to see if I have to charge them.

### Headset Name

SteelSeries Arctis Nova 7

### On which OS does the problem happen?

Linux

### Device information

<details>
 <summary>Detailed Device Information</summary>

 ```
Device Found
  VendorID: 0x1038
 ProductID: 0x2202
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: SteelSeries
  Product:      Arctis Nova 7
  Interface:    3
  Usage-Page: 0xffc0 Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x2202
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: SteelSeries
  Product:      Arctis Nova 7
  Interface:    4
  Usage-Page: 0xc Usageid: 0x1

Device Found
  VendorID: 0x1038
 ProductID: 0x2202
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: SteelSeries
  Product:      Arctis Nova 7
  Interface:    5
  Usage-Page: 0xff00 Usageid: 0x1
 ```

</details>

nicola02nb, 2024-10-01:
Did you build headsetcontrol by yourself?

Because I edited Artic Nova 7 Wireless code a few months ago and putted wrong numbers to for the different battery levels by following user manual, and then rolled back original code for that because older one was more inline with Steelseries Engine advertised battery level.

I also observerd that hadphones led shows completely different levels than the values of Steelseries Engine.

If you anlyze headphones packets they follow Steelseries Engine levels:
| Percentage | Packet Code |
|--|--|
| 100 | 04 | 
| 75 | 03 |  
| 50 | 02 | 
| 25 | 01 |  
| 0 (Never reached) | 00 (Probably) |

While instead into headphones manual the reported levels are:
| Percentage | Color |
|--|--|
| 100-50 | Green | 
| 49-15 | Yellow |  
| 14-5 | Red | 
| 4-1 | Red Fast Blink |

ProLejka, 2024-10-01:
Ah... I did and I probably hit the time when it was like that HeadsetControl (3.0.0-17-gb4d0614)
Now on HeadsetControl (3.0.0-23-gc0c9b50) it looks fine. Sorry in future I'll remeber to pull first.
Thank you.

closed 2024-10-01 by ProLejka

nicola02nb, 2024-10-02:
No worries, it was also my fault to edit code that was already correct