Sapd/HeadsetControl #374: [gentoo overlay] no ebuilds to satisfy "dev-utils/cmake"

boprr, 2024-10-05:
### Description

The package named `dev-utils/cmake` is incorrect; the correct package is `dev-build/cmake`.

### On which OS does the problem happen?

Linux

mattmcadoo, 2024-12-06:
That overlay is super out of date and the last ebuilds for headsetcontrol were updated 3 years ago.

mattmcadoo, 2024-12-06:
Because of the error I was getting that was the same as noted in #335  I created my own ebuild for 3.0.0 that compiles for me:

```
EAPI=8

inherit cmake git-r3 udev

DESCRIPTION="Adds loopback and LED control to headsets"
HOMEPAGE="https://github.com/Sapd/HeadsetControl"
SRC_URI=""
EGIT_REPO_URI="https://github.com/Sapd/HeadsetControl"

if [[ ${PV} = 9999 ]] ; then
	EGIT_COMMIT=""
else
	EGIT_COMMIT="${PV}"
	KEYWORDS="~amd64 ~x86"
fi

LICENSE="GPL-3"
SLOT="0"
IUSE="udev"

DEPEND="
	dev-build/cmake
	dev-libs/hidapi
	udev? ( virtual/udev )
"
RDEPEND="
	dev-libs/hidapi
"
BDEPEND=""

CMAKE_MAKEFILE_GENERATOR="emake"

src_prepare() {
	default
	sed -i 's/set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -Werror")//g' "${WORKDIR}/${P}/CMakeLists.txt" || die "Failed to remove Cmake flags"
	sed -i "s;lib/udev/rules.d;$(get_udevdir)/rules.d;" "${WORKDIR}/${P}/CMakeLists.txt" || die "Failed to replace the udev rules location"
	cmake_src_prepare
}

pkg_postinst() {
	if use udev; then
		udev_reload
	fi
}
```

boprr, 2024-12-12:
o **** forgot about this (thanks for the ebuild tho)